KonradIT/gopro-py-api #1: Can't install the API, setup.py missing

arthurpattee, 2017-03-12:
Hi ! I'm trying to install your library however I'm getting an error since there is no setup.py file :/

KonradIT, 2017-03-12:
Hi, I know, right now just use the commands:

```
from goprocam import GoProCamera
from goprocam import constants
gopro = GoProCamera.GoPro() #recommended: add constants.gpcontrol for HERO4/5 or constants.auth for HERO3/3+

...
```

The setup will be added later on.

closed 2017-03-12 by KonradIT

arthurpattee, 2017-03-12:
New error, when trying to import GoProCamera returns 'No module named request' (is there a library to install? I do babe urllib running though )

KonradIT, 2017-03-12:
@arthurpattee make sure you're running python 3.

arthurpattee, 2017-03-12:
@KonradIT I was using python 2.7, seems to be better with 3.0 indeed, yet still an error when lancing gopro = GoProCamera.GoPro()
-> the JSON object must be str, not 'bytes'

KonradIT, 2017-03-12:
@arthurpattee I'm using Python 3.6.0.

What camera are you using?

arthurpattee, 2017-03-12:
@KonradIT I'm using python 3.5.2 and I've tried with a Hero 5 Session and a hero 4 black, same respond on both

KonradIT, 2017-03-12:
@arthurpattee 

Then use: 
    
    gopro = GoProCamera.GoPro(constants.gpcontrol)

arthurpattee, 2017-03-12:
@KonradIT Same error, but at slightly different line I noticed, it starts with a Waking up... message then the JSON object..

KonradIT, 2017-03-12:
@arthurpattee just pushed a fix, hope it works. Please pull the latest changes and try again. Also, tell me which line exactly.

arthurpattee, 2017-03-12:
@KonradIT it works ! Thank you very much! If I can do something to help debug let me know! 😊