KonradIT/gopro-py-api #2: Issue while trying to take picture

arthurpattee, 2017-03-18:
Hi there! Your api was working great last week, but I tried to use it on my new PI Zero W and to seems not to be working anymore, I get the error : ´str' does not support the buffer interface in line 96

KonradIT, 2017-03-19:
Thanks for the bug report, fixed in 867b2a4

arthurpattee, 2017-03-19:
@KonradIT Not the same mistake but still, while trying 'gopro = GoProCamera.GoPro(constants.gpcontrol)' returns Waking up... and then get locked on that state

KonradIT, 2017-03-19:
@arthurpattee what camera?

arthurpattee, 2017-03-19:
@KonradIT GoPro hero 5 session :)

cavenel, 2017-03-19:
That looks more like a WiFi problem, are you sure the PI Zero W is on the GoPro WiFi? What is the result of ```ifconfig``` on the PI Zero W?

arthurpattee, 2017-03-19:
@cavenel Hi, yes i'm sure it is, ifconfig returns:
pi@raspberrypi:~/gopro-py-api $ ifconfig
lo        Link encap:Boucle locale  
          inet adr:127.0.0.1  Masque:255.0.0.0
          adr inet6: ::1/128 Scope:Hôte
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:143 errors:0 dropped:0 overruns:0 frame:0
          TX packets:143 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 lg file transmission:1 
          RX bytes:12233 (11.9 KiB)  TX bytes:12233 (11.9 KiB)

wlan0     Link encap:Ethernet  HWaddr b8:27:eb:e2:0e:fc  
          inet adr:10.5.5.100  Bcast:10.5.5.255  Masque:255.255.255.0
          adr inet6: fe80::4718:b81d:49f:db97/64 Scope:Lien
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:24327 errors:0 dropped:5850 overruns:0 frame:0
          TX packets:7675 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 lg file transmission:1000 
          RX bytes:23975773 (22.8 MiB)  TX bytes:890040 (869.1 KiB)

KonradIT, 2017-03-19:
you may want to set a custom Mac address in the GoPro() field as well as constants.gpcontrol

KonradIT, 2017-03-19:
I'll try with my session now (Hero4 session not 5)

arthurpattee, 2017-03-19:
@KonradIT cheers for the support :)

arthurpattee, 2017-03-19:
@KonradIT Like that ? gopro = GoProCamera.GoPro(F6:DD:9E:7A:etc.constants.gpcontrol) ?

KonradIT, 2017-03-19:
yeah, but between " " and with a , at the end diving from constants.gpcontrol

KonradIT, 2017-03-19:
@arthurpattee, this may be because HERO5 use some sort of different WoL command: https://github.com/KonradIT/goprowifihack/issues/65

arthurpattee, 2017-03-19:
@KonradIT first step seems to be working  
gopro = GoProCamera.GoPro("F6:DD:9E:7A:A1:93",constants.gpcontrol)
Connected to 10.5.5.9

but then:
>>> gopro.downloadLastMedia(gopro.take_photo())
Waking up...
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/gopro-py-api/goprocam/GoProCamera.py", line 389, in take_photo
    self.mode(constants.Mode.PhotoMode)
  File "/home/pi/gopro-py-api/goprocam/GoProCamera.py", line 294, in mode
    if self.whichCam() == "gpcontrol":
  File "/home/pi/gopro-py-api/goprocam/GoProCamera.py", line 173, in whichCam
    self.power_on(self._mac_address)
  File "/home/pi/gopro-py-api/goprocam/GoProCamera.py", line 345, in power_on
    raise ValueError('Incorrect MAC address format')
ValueError: Incorrect MAC address format

arthurpattee, 2017-03-19:
@KonradIT working for me with the 5 session:
pi@raspberrypi:~/gopro-py-api $ wakeonlan -i 10.5.5.9 F6:DD:9E:7A:A1:93
Sending magic packet to 10.5.5.9:9 with F6:DD:9E:7A:A1:93

KonradIT, 2017-03-19:
great!!! I'll fix that now. Thanks for this report man.

arthurpattee, 2017-03-19:
@KonradIT Ahah no problem glad I can return a bit of the help you provided me ;)

KonradIT, 2017-03-19:
hey @arthurpattee, you could first wake the camera with the command wakeonlan and then use the API.

arthurpattee, 2017-03-19:
@KonradIT That's what I fought but when I try to define the gopro variable with gopro = GoProCamera.GoPro(constants.gpcontrol)
It returns 'Waking up..' even if the cam is up :/

KonradIT, 2017-03-19:
it's very strange. try searching for wake on lan python snippets for Python 3 and see which one works, sorry it's too much but without a camera here is hard to debug.

arthurpattee, 2017-03-19:
@KonradIT no worries I understand and appreciate your help !
Before I was using your api I was using the os librarie and a os.system("wakeonlan..") oldschool but working :p

KonradIT, 2017-03-19:
I'll try and send snippets of python code over here for you to try, I'd rather put a native python solution instead of calling a system command. I want to be as OS-agnostic as I can. I'm also gonna get a HERO5 Black in a couple of days.

arthurpattee, 2017-03-19:
@KonradIT cheers!
I understand, if you find something let me know ;)
Ahah good excuse to get the new one ;)

KonradIT, 2017-03-19:
yup, been since 2014 with the 4 black and later on with the session, time to upgrade!

KonradIT, 2017-03-19:
Hey @arthurpattee 

Try this out: http://www.sgvulcan.com/2010/01/16/wake-on-lan-a-machine-using-python-wol/

I can also wake my HERO4 Black with the already exisiting code: https://vimeo.com/209079783 (starts at minute 15)

KonradIT, 2017-03-19:
hey @arthurpattee I've identified a bug I'm working on a release, stay tuned

KonradIT, 2017-03-19:
Hey @arthurpattee , just fixed and pushed to pypi and github, please update pypi or pull from github and see if it works.

arthurpattee, 2017-03-21:
Hey there ! I downloaded the latest version from github however I get the same error, camera being up or not..
Locked on Waking up.. :/

arthurpattee, 2017-03-21:
@KonradIT However, if you can explain me how to skip this step, because I got the solution to wake up my gopro (with my os.system command :p ), I would be able to use the script smoothly

KonradIT, 2017-03-21:
@arthurpattee hmm.. strange, could you include a screenshot?

arthurpattee, 2017-03-21:
@KonradIT Working on the PI Zero W, so a bit hard to screenshot, I took photos, hope you can read well

[image]
[image]

KonradIT, 2017-03-21:
@arthurpattee Let's do 2 tests:

Uninstall any version from PyPi, use it from the directory.

Test 1) Remove lines 28 to 36 in GoProCamera.py

Test 2) Comment any ```self.power_on(``` you see

arthurpattee, 2017-03-21:
@KonradIT So I did launch python from the fresh download directorie I've made, then tried out both of the test (I didn't replace the 28-36 lines for the second one, do I have to?) And same result.
I've waited about 3 min each time but no response.. :/

arthurpattee, 2017-03-21:
Test 1 : [image]
Test 2 : [image]

KonradIT, 2017-03-21:
I think I got it, in line 28 of the Python script remove ```"HD5" in response or ``` and it should work.

I have my HERO4 Session booted up in App mode from previous tests and when I connect it gives me no issues.
[image]

KonradIT, 2017-03-21:
Maybe try updating Python to 3.6.0?? Also, clone my repo again before doing anything just to make sure we're on the same page. 

BTW the seller of the H5 Black I was going to buy already sold it (it was 260EUR which is an incredible deal) so I might end up buying a H5 Session

arthurpattee, 2017-03-21:
@KonradIT I removed the mention, tried and failed, rebooting the Pi and will retry in few seconds
This is clearly weird..
I will try updating my python version
Damn that was a great deal indeed! I'm prototyping on it but if I may, I would suggest you to still buy the 5 black, the 5 session seems to be just a 4 black in a smaller body ;)

KonradIT, 2017-03-21:
@arthurpattee yeah, will see on eBay.

Another way to see what is going on is to create a python script solely to wake the camera on. 

Create a python script with this contents inside:

```
import socket
import struct

def wake_on_lan(macaddress):
    """ Switches on remote computers using WOL. """

    # Check macaddress format and try to compensate.
    if len(macaddress) == 12:
        pass
    elif len(macaddress) == 12 + 5:
        sep = macaddress[2]
        macaddress = macaddress.replace(sep, '')
    else:
        raise ValueError('Incorrect MAC address format')
 
    # Pad the synchronization stream.
    data = ''.join(['FFFFFFFFFFFF', macaddress * 20])
    send_data = '' 

    # Split up the hex values and pack.
    for i in range(0, len(data), 2):
        send_data = ''.join([send_data,
                             struct.pack('B', int(data[i: i + 2], 16))])

    # Broadcast it to the LAN.
    sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
    sock.setsockopt(socket.SOL_SOCKET, socket.SO_BROADCAST, 1)
    sock.sendto(send_data, ('', 7))

wake_on_lan('F6DD9E7AA193')
#F6:DD:9E:7A:A1:93
```

arthurpattee, 2017-03-21:
Ok so downloaded the latest release from you github, cd /directory then python 3 and typed the same commands you did, but still :
[image]
I don't understand that beast :( How can I incorporate your script into the librairie?

arthurpattee, 2017-03-21:
@KonradIT Ok so I finally decided to reset the wifi connection on my gopro and everything seems to work just fine, it's so weird!

KonradIT, 2017-03-21:
What? Is it working? What code is working?

arthurpattee, 2017-03-21:
Yep, it's working, with your new library and my old script :p Thanks for the support! That means the Gopro is kinda bugging after all, we could have anticipated it ahah

KonradIT, 2017-03-21:
@arthurpattee what code works after wifi connection reset? [This comment](https://github.com/KonradIT/gopro-py-api/issues/2#issuecomment-288213664) or the library?

KonradIT, 2017-03-21:
@arthurpattee did you modify the library? I really want to know if you can wake up the camera using Python not the command ```wakeonlan```

KonradIT, 2017-03-23:
hey @arthurpattee can you test the scripts?

arthurpattee, 2017-03-23:
@KonradIT I will try to sum up my experience in that one later on today, got a lot of work and not much time for my gopro project :/

KonradIT, 2017-03-23:
@arthurpattee no worries, I just want to know if you can turn on a H5 Session with python only

iayanpahwa, 2017-07-03:
Hi, I've a GoPro Hero 5 session cam and had similar issues, in case you patched and need to test it, please feel to reach out to me. I have python programming experience and could help you test it on H5 session, currently can wake cam using wakeonlan command, will test other things tomorrow :)

KonradIT, 2017-07-04:
@iayanpahwa don't have the H5 yet, what's exactly not working?

iayanpahwa, 2017-07-05:
@KonradIT not able to switch on the cam.
Host : OSX python3, raspberry pi Jessie python3
It hangs after running gpcam = GoProcamera.GoPro() in interactive session.
Tried with MAC address as well and arguments specific to H5 cameras. If I manually turn it on the hang state die spitting out exceptions on stdout. Although I am able to turn it on using wakeonlan.

iayanpahwa, 2017-07-19:
Hi @KonradIT any updates on same?

KonradIT, 2017-07-19:
I would need a HERO5 camera to debug it, a user found out it is due to my connection check, but I want to have a connection check + while connecting to H5.

KonradIT, 2017-08-10:
@arthurpattee @iayanpahwa I'm investigating the issue where I cannot turn on my HERO5 Black without specifying my own MAC address. the fix is to specify the Mac address in GoPro(). You can also leave it blank and use mine but in case that does not work you can specify your own address and it will work. I will also be fixing other H5 issues. if you have any other issue let me know.