KonradIT/gopro-py-api #3: urllib.request

isdito, 2017-03-28:
hello,

Thank you for the post before.

What version of python need ? 

```
    import urllib.request
ImportError: No module named request
```
I have version 2.7 I need install another verison

KonradIT, 2017-03-28:
you need python 3.6

isdito, 2017-03-28:
OK I install. I go to check.
I need capture at 10 minutes and transfer to my computer it is more easy but I have more problems with another programs like Action camera tool box

isdito, 2017-03-28:
The gpCam.downloadLastMedia() I could change the path to download, I dont know the path.

Thank you

isdito, 2017-03-28:
OK I find it Current directory. I think that it is better selec a foder
import os
os.getcwd()

KonradIT, 2017-03-28:
hi, as per the custom path for download:

gpCam.downloadLastMedia(custom_filename="/home/blahblah/Videos/GoPro/")

closed 2017-06-13 by KonradIT