KonradIT/gopro-py-api #4: power on and off

isdito, 2017-03-28:
Hello.
If you dont mind clarify me.

it is posible power on at 7 am and power off at 7pm, day after day and not lost the connection with a computer.

Also I could change values like protune, flat or exposition in real time.

Thank you

KonradIT, 2017-03-28:
Hi,

what camera do you have?

for power on the camera at 7am and power off at 7pm, you will need a battery eliminator or a USB cable connected to a socket. Keep in mind GoPro cameras tend to shut down or turn off the wifi because of heat.

```
from datetime import datetime
from threading import Timer
x=datetime.today()
morning=x.replace(day=x.day, hour=7, minute=0, second=0, microsecond=0)
afternoon=x.replace(day=x.day, hour=19, minute=0, second=0, microsecond=0)
delta_m=y-morning
delta_a=y-afternoon
secs_m=delta_m.seconds
secs_a = delta_m.seconds

morning_timer = Timer(secs_m, gpCam.power_on())
morning_timer.start()
afternoon_timer = Timer(secs_a, gpCam.power_off())
afternoon_timer.start()
```

haven't tested the code
```

isdito, 2017-03-28:
hello.

I create a while() and inside I create with datime... This works. The problem it is powe_on and off

HERO 5

the problem is that gpCam.power_off()  disable de camera but the power_on() not enable de camera

KonradIT, 2017-03-28:
yeah, we cannot seem to power on the HERO5 from python, see #2

isdito, 2017-03-28:
I think that better not put power off ;). I must to put the camera at 25 meters tall and one year.

I hope not stop o break with 1 frame in 10 minutes 7 am at 7 pm.

You think that could lost wiffy connection with the computer? more times and not recover. You think that there are other machines better like Saomin..

Thank you

KonradIT, 2017-03-28:
I personally wouldn't recommend a HERO5 for long timelapses,at least not right now. Buy a HERO3 and use [MegaLapse](http://chernowii.com/MegaLapse) hack.

isdito, 2017-03-28:
OK Thank you I revise. The porblem is that I buy a hero 5 ;) but I go to see the web that you send me.

isdito, 2017-03-29:
Hello.

Yes you have reason the go pro HERO5 freeze wiffi networks when you are asking 40 times... very bad.
I execute "netsh wlan connect name=GP2...." to refresh wiffy connection but the camera lost the control

Konradlt it is posible execute a time laps not record a photo o video , send that execute a time laps and stop it and in background ask frames to the go pro in the intervall.

I see the constants but I dont know how it works.

Sorry my bad english.

KonradIT, 2017-03-29:
Habla en español si no. puedes conectarte a la GoPro desde la desktop environment?

isdito, 2017-03-29:
Buenas tardes,
 ahhh vaya Gracias. 

No lo tengo instalado el capture. Ahora he anulado lo que es la transferencia para saber en que punto se atasca la gopro si es por mandar el comando o por ejecutar traerte la imagen.

Ahora te pongo el codigo

KonradIT, 2017-03-29:
```
import datetime
from goprocam import GoProCamera
import time
while(1):
	now = datetime.datetime.now()
	now_time = now.time()
	print (now_time)
	if now_time >= datetime.time(7,00) and now_time <= datetime.time(22,30):
		gpCam = GoProCamera.GoPro()
		gpCam.power_on()
		f=gpCam.downloadLastMedia(gpCam.take_photo(3))
		archivo=gpCam.getMediaInfo("folder") + "-" + gpCam.getMediaInfo("file")
		print("Nombre del archivo: " + archivo)
		#subir_a_ftp(archivo)
	else:
		if(encendida==0): 
			endendida=1
			gpCam.power_off()
	print ("Estamos esperando al siguiente frame")
	time.sleep(3)
def subir_a_ftp(f):
	FTP_salida("ftp.video-eventos.com","stepark","******",f)
```

Ese funciona con mi HERO4 Black.

isdito, 2017-03-29:
Pero gpCam.power_on() gpCam.power_off()  no funciona con la Hero 5 ¿no?

KonradIT, 2017-03-29:
tienes que actualizar el repositorio. Descargarlo otra vez o usa pip install goprocam

isdito, 2017-03-29:
Si ya lo he realizado he dejado ahora lago pro a ver si se bloquea

-- 

Israel D.
Color Grading , Editor , shooter

www.video-boda.es
israel@video-boda.es
Tef.:  +34 654 43 83 39

La información contenida en este correo electrónico, y en su caso,
cualquier fichero anexo al mismo, son de carácter privado y confidencial
siendo para uso exclusivo de su destinatario. Si usted no es el
destinatario correcto, el empleado o agente responsable de entregar el
mensaje al destinatario, o ha recibido esta comunicación por error, le
informamos que está totalmente prohibida cualquier divulgación,
distribución o reproducción de esta comunicación según la legislación
vigente y le rogamos que nos lo notifique inmediatamente, procediendo a su
destrucción sin continuar su lectura.  Le informamos que su dirección de
correo electrónico, así como el resto de los datos de carácter personal de
la tarjeta de visita que nos facilite, podrían ser objeto de tratamiento
automatizado en nuestros ficheros, con la finalidad de gestionar la agenda
de contactos de nuestra empresa. Vd. podrá en cualquier momento ejercer el
derecho de acceso, rectificación, cancelación y oposición en los términos
establecidos en la Ley Orgánica 15/1999 mediante notificación escrita a la
siguiente dirección Calle Rio Duero 72 , San Antonio de benageber , 46184,
Valencia.

KonradIT, 2017-03-29:
si lo has actualizado no te debería dar error el power on

KonradIT, 2017-03-29:
en la actualizacion 1.4 funciona el power_off() en la HERO5. Prueba si funciona el power_on() por que no se.

Aqui hay ejemplos de timelapses que descargan las fotos.

https://github.com/KonradIT/gopro-py-api/blob/master/examples/download_timelapse.py toma una foto, la descarga y se repite (cambia el valor de TIMER a 30)

https://github.com/KonradIT/gopro-py-api/blob/master/examples/stored_timelapse.py toma una foto, la descarga, la borra y se repite (tambien cambia el TIMER a 30)

isdito, 2017-03-29:
Buenas noches,
Perdona había salido a cenar, he dejado la go pro para ver si mañana se enciende sola. Ya te comento.

Me gustaría ver si tienes un ejemplo de programar en ver de modo photo, modo timelaps con un tiempo determinado. Si tuvieras un ejemplo del compando.

Un saludo.

KonradIT, 2017-04-01:
Hola @isdito 

>El comando que pasa a modo timelaps.

```python
gpCam.mode(constants.Mode.Multishot, constants.Mode.Submode.TimeLapse)
```

>El comando que marca segundos entre exposicion.

Para timelapse:

```python
gpCam.gpControlSet(constants.Multishot.TIMELAPSE_INTERVAL, constants.Multishot.TimeLapseInterval.I10)
```

.I10 es in intervalo de 10 segundos. Puedes ponder I1, I2, I5, I10, I30, I60 o IHalfSec para 0.5 segundos.

>El comando que controla el iso, flat, hdr,raw

ISO de que? La HERO5 Black soporta ISO para video, foto y multishot. Para video:

```python
gpCam.gpControlSet(constants.Video.ISO_LIMIT, constants.Video.IsoLimit.ISO1600)
```
por ejemplo.

Mira el constants.py para ver lo que puedes poner.

Para el flat seguro que es el modo de color:

```python
gpCam.gpControlSet(constants.Video.COLOR, constants.Video.Color.Flat)
```

>El comando que hace play

Que? no hay ningun comando que haga que se reproduzca el video en la pantalla.

>y el comando que para.
que para el video?

```python
gpCam.shutter(constants.stop)
```

Lee el README.md y constants.py para entender la libreria.

isdito, 2017-04-01:
> Hola @isdito <https://github.com/isdito>
>
> El comando que pasa a modo timelaps.
>
> gpCam.mode(constants.Mode.Multishot, constants.Mode.Submode.TimeLapse)
>
> El comando que marca segundos entre exposicion.
>
> Para timelapse:
>
> gpCam.gpControlSet(constants.Multishot.TIMELAPSE_INTERVAL, constants.Multishot.TimeLapseInterval.I10)
>
> .I10 es in intervalo de 10 segundos. Puedes ponder I1, I2, I5, I10, I30,
> I60 o IHalfSec para 0.5 segundos.
>
> El comando que controla el iso, flat, hdr,raw
>
> ISO de que? La HERO5 Black soporta ISO para video, foto y multishot. Para
> video:
>
> gpCam.gpControlSet(constants.Video.ISO_LIMIT, constants.Video.IsoLimit.ISO1600)
>
> por ejemplo.
>
> Mira el constants.py para ver lo que puedes poner.
>
> Para el flat seguro que es el modo de color:
>
> gpCam.gpControlSet(constants.Video.COLOR, constants.Video.Color.Flat)
>
> El comando que hace play
>
> Que? no hay ningun comando que haga que se reproduzca el video en la
> pantalla.
>
> y el comando que para.
> que para el video?
>
> gpCam.shutter(constants.stop)
>
> Lee el README.md y constants.py para entender la libreria.

muchas graciaspor la explicación la cámara no consigo que se encienda con
él porwer on, Y acaba bloqueándose y perdiendo la conexión con el paso de
las horas.

 Un saludo

-- 

Israel D.
Color Grading , Editor , shooter

www.video-boda.es
israel@video-boda.es
Tef.:  +34 654 43 83 39

La información contenida en este correo electrónico, y en su caso,
cualquier fichero anexo al mismo, son de carácter privado y confidencial
siendo para uso exclusivo de su destinatario. Si usted no es el
destinatario correcto, el empleado o agente responsable de entregar el
mensaje al destinatario, o ha recibido esta comunicación por error, le
informamos que está totalmente prohibida cualquier divulgación,
distribución o reproducción de esta comunicación según la legislación
vigente y le rogamos que nos lo notifique inmediatamente, procediendo a su
destrucción sin continuar su lectura.  Le informamos que su dirección de
correo electrónico, así como el resto de los datos de carácter personal de
la tarjeta de visita que nos facilite, podrían ser objeto de tratamiento
automatizado en nuestros ficheros, con la finalidad de gestionar la agenda
de contactos de nuestra empresa. Vd. podrá en cualquier momento ejercer el
derecho de acceso, rectificación, cancelación y oposición en los términos
establecidos en la Ley Orgánica 15/1999 mediante notificación escrita a la
siguiente dirección Calle Rio Duero 72 , San Antonio de benageber , 46184,
Valencia.

KonradIT, 2017-04-01:
Hola, como he mencionado antes no la enciendas. Conectala ya encendida.

isdito, 2017-04-01:
Si pero konrad si la apagas se acabo
 Gracias por todo

-- 

Israel D.
Color Grading , Editor , shooter

www.video-boda.es
israel@video-boda.es
Tef.:  +34 654 43 83 39

La información contenida en este correo electrónico, y en su caso,
cualquier fichero anexo al mismo, son de carácter privado y confidencial
siendo para uso exclusivo de su destinatario. Si usted no es el
destinatario correcto, el empleado o agente responsable de entregar el
mensaje al destinatario, o ha recibido esta comunicación por error, le
informamos que está totalmente prohibida cualquier divulgación,
distribución o reproducción de esta comunicación según la legislación
vigente y le rogamos que nos lo notifique inmediatamente, procediendo a su
destrucción sin continuar su lectura.  Le informamos que su dirección de
correo electrónico, así como el resto de los datos de carácter personal de
la tarjeta de visita que nos facilite, podrían ser objeto de tratamiento
automatizado en nuestros ficheros, con la finalidad de gestionar la agenda
de contactos de nuestra empresa. Vd. podrá en cualquier momento ejercer el
derecho de acceso, rectificación, cancelación y oposición en los términos
establecidos en la Ley Orgánica 15/1999 mediante notificación escrita a la
siguiente dirección Calle Rio Duero 72 , San Antonio de benageber , 46184,
Valencia.

KonradIT, 2017-04-01:
Puedes encenderla con ```wakeonlan``` para Debian/Ubuntu, mira si hay algun comando para windows.

isdito, 2017-04-01:
https://pypi.python.org/pypi/wakeonlan/0.2.2

¿Con esta librería ? Con la ip. Saludos

-- 

Israel D.
Color Grading , Editor , shooter

www.video-boda.es
israel@video-boda.es
Tef.:  +34 654 43 83 39

La información contenida en este correo electrónico, y en su caso,
cualquier fichero anexo al mismo, son de carácter privado y confidencial
siendo para uso exclusivo de su destinatario. Si usted no es el
destinatario correcto, el empleado o agente responsable de entregar el
mensaje al destinatario, o ha recibido esta comunicación por error, le
informamos que está totalmente prohibida cualquier divulgación,
distribución o reproducción de esta comunicación según la legislación
vigente y le rogamos que nos lo notifique inmediatamente, procediendo a su
destrucción sin continuar su lectura.  Le informamos que su dirección de
correo electrónico, así como el resto de los datos de carácter personal de
la tarjeta de visita que nos facilite, podrían ser objeto de tratamiento
automatizado en nuestros ficheros, con la finalidad de gestionar la agenda
de contactos de nuestra empresa. Vd. podrá en cualquier momento ejercer el
derecho de acceso, rectificación, cancelación y oposición en los términos
establecidos en la Ley Orgánica 15/1999 mediante notificación escrita a la
siguiente dirección Calle Rio Duero 72 , San Antonio de benageber , 46184,
Valencia.

KonradIT, 2017-04-01:
```wol.send_magic_packet('ff.ff.ff.ff.ff.ff', ip_address='10.5.5.9', port=9)```

prueba eso (si no funciona reemplaza la Mac address por la de tu camara. Para conseguir tu Mac address ve a 10.5.5.9/gp/gpControl/info en un navegador y estara ahi.

isdito, 2017-04-01:
Thank you for all. Chequeo todo y reviso .

-- 

Israel D.
Color Grading , Editor , shooter

www.video-boda.es
israel@video-boda.es
Tef.:  +34 654 43 83 39

La información contenida en este correo electrónico, y en su caso,
cualquier fichero anexo al mismo, son de carácter privado y confidencial
siendo para uso exclusivo de su destinatario. Si usted no es el
destinatario correcto, el empleado o agente responsable de entregar el
mensaje al destinatario, o ha recibido esta comunicación por error, le
informamos que está totalmente prohibida cualquier divulgación,
distribución o reproducción de esta comunicación según la legislación
vigente y le rogamos que nos lo notifique inmediatamente, procediendo a su
destrucción sin continuar su lectura.  Le informamos que su dirección de
correo electrónico, así como el resto de los datos de carácter personal de
la tarjeta de visita que nos facilite, podrían ser objeto de tratamiento
automatizado en nuestros ficheros, con la finalidad de gestionar la agenda
de contactos de nuestra empresa. Vd. podrá en cualquier momento ejercer el
derecho de acceso, rectificación, cancelación y oposición en los términos
establecidos en la Ley Orgánica 15/1999 mediante notificación escrita a la
siguiente dirección Calle Rio Duero 72 , San Antonio de benageber , 46184,
Valencia.

KonradIT, 2017-04-07:
Hola @isdito 

He actualizado la API, ahora deberia funcionar con al GoPro HERO5 (incluido el encendido de la camara)

isdito, 2017-04-07:
Gracias KonradIT .

Ya lo instalo y te comento. Saludos.

G0ldDiggAA, 2017-04-28:
Hello !
Do you update the lib @KonradIT ? it seems to doesn't work the power_on() for the Hero 5.
The power_off() works great, but the gopro doesn't wake up with the power_on() command ...
Any idea ?

Thanks for your great job !

KonradIT, 2017-04-28:
hi @G0ldDiggAA, I do not have a H5 so I cannot test but if you want I can send you test scripts and you can test which one turns the camera on. Also a Wireshark pcap would help

isdito, 2017-04-28:
hello if you need I test

Send me the script And test for you

Israel D.
Color Grading , Editor , shooter

www.video-boda.es
israel@video-boda.es
Tef.:  +34 654 43 83 39

La información contenida en este correo electrónico, y en su caso,
cualquier fichero anexo al mismo, son de carácter privado y confidencial
siendo para uso exclusivo de su destinatario. Si usted no es el
destinatario correcto, el empleado o agente responsable de entregar el
mensaje al destinatario, o ha recibido esta comunicación por error, le
informamos que está totalmente prohibida cualquier divulgación,
distribución o reproducción de esta comunicación según la legislación
vigente y le rogamos que nos lo notifique inmediatamente, procediendo a su
destrucción sin continuar su lectura.  Le informamos que su dirección de
correo electrónico, así como el resto de los datos de carácter personal de
la tarjeta de visita que nos facilite, podrían ser objeto de tratamiento
automatizado en nuestros ficheros, con la finalidad de gestionar la agenda
de contactos de nuestra empresa. Vd. podrá en cualquier momento ejercer el
derecho de acceso, rectificación, cancelación y oposición en los términos
establecidos en la Ley Orgánica 15/1999 mediante notificación escrita a la
siguiente dirección Calle Rio Duero 72 , San Antonio de benageber , 46184,
Valencia.

G0ldDiggAA, 2017-04-28:
@KonradIT yes send me your tests scripts and i'm gonna give you my feedback. And thanks @isdito for your help about this. It's better to be two than one !

KonradIT, 2017-04-28:
hey all .
here are the test scripts: https://gist.github.com/KonradIT/d150c671e25b01c49e998ad8d5ca3e1a

how to test:

camera must be updated to the newest firmware, powered off and with WiFi on. The computer must be connected to the camera WiFi.

Run each script, wait 10 seconds to see if the camera turns on, if not run the script again for a second time (sometimes it needs this), if it does not power on then go ahead with the next script

KonradIT, 2017-04-28:
note: script 2 needs wakeonlan library from pip, install using ```pip install wakeonlan```