KonradIT/gopro-py-api #5: Problem with MultiShot Mode and Download

Teo47, 2017-05-15:
Hi again Konrad !
I'm always on my photogrammetry project. I used your code to create an interface on Qt, it is working perfectly, such a good work !!!

I have just one problem. I need to download photos and rename file with the timestamp. No problem for rename a folder with your example, the problem is in downloading photos. I need to use MultiShot Mode and it's only downloading the first photo of the MultiShot. I tryied with DowloadAll and it is the same problem.

Do you have any issue to get infos of all MultiShot photos ?
I join my code with definition of my Qt fonctions.

```
 def connexion(self):
        self.gpCam = GoProCamera.GoPro()
        if self.gpCam.getStatusRaw()== None:
            self.actual_statut_2.setText("Erreur")
        else:
            self.actual_statut_2.setText("Connecté")

    def Allumer(self):
        self.gpCam.power_on()

    def Eteindre(self):
        self.gpCam.power_off()

    def play(self):
        self.actual_statut.setText("PLAY")
        self.gpCam.shutter(constants.start)
        if self.gpCam.getStatus(constants.Status.Status,constants.Status.STATUS.IsBusy) == 1:
            self.debut = datetime.datetime.utcnow()
            print(self.debut)

    def download(self):
        self.actual_statut.setText("DOWNLOAD")
        media = self.gpCam.listMedia(True, True)
        self.gpCam.downloadAll()

    def stop(self):
        self.actual_statut.setText("STOP")
        self.gpCam.shutter(constants.stop)
        self.fin = datetime.datetime.utcnow()
        print(self.fin)
        self.delta_t = self.fin - self.debut
        print(self.delta_t)

    def Timelapse05(self):
        self.actual_statut.setText("05")
        self.gpCam.mode(constants.Mode.MultiShotMode, constants.Mode.SubMode.MultiShot.TimeLapse)
        self.gpCam.gpControlSet(constants.Multishot.TIMELAPSE_INTERVAL, constants.Multishot.TimeLapseInterval.IHalf1)

    def Timelapse1(self):
        self.actual_statut.setText("1")
        self.gpCam.mode(constants.Mode.MultiShotMode, constants.Mode.SubMode.MultiShot.TimeLapse)
        self.gpCam.gpControlSet(constants.Multishot.TIMELAPSE_INTERVAL, constants.Multishot.TimeLapseInterval.I1)
```

KonradIT, 2017-05-15:
haven't looked at downloading a batch, need to finish some exams first so I'll do that at the start of June.

Teo47, 2017-06-12:
Up ! Did you have time to check my issues ?

KonradIT, 2017-06-12:
yes I have it solved

KonradIT, 2017-06-12:
https://github.com/KonradIT/gopro-py-api/blob/master/goprocam/GoProCamera.py#L546

closed 2017-06-13 by KonradIT