KonradIT/gopro-py-api #7: Using getMediaInfo inside a script

joaolopes220, 2017-06-12:
Hello,
Thanks for the library, it's very helpful.
I'm running into one problem, and I'm not really sure where to look at to solve.
When I run the functions as a command it works fine, and most of the functions work fine in a script as well.
However, when I run getMediaInfo inside a script it always returns an empty string. 

Thank you again

KonradIT, 2017-06-13:
Hi.
Can you post the script? I think is because ```.getMediaInfo()``` ```returns``` a string, not ```print```, ```return``` works in the python REPL but not when running ```py script.py```. Try encapsulating your getMediaInfo line inside a ```print()``` statement.
[image]

closed 2017-06-13 by KonradIT

joaolopes220, 2017-06-13:
Hello, sorry it took a while to get back.  
If I run something simple like this, I don't need to print the file name, I'm just trying to store it.  
If I run this then it fails to store it and gpfile is saved as an empty string.  I'm trying to get the file name saved so I can better track the videos, but also to use it to get the information of the hilight tags.  

gpCam = GoProCamera.GoPro()
gpCam.syncTime()
gpCam.shutter(constants.start)
time.sleep(5)
gpCam.shutter(constants.stop)
gpfile = gpCam.getMediaInfo('file')

KonradIT, 2017-06-13:
@joaolopes220 using ```print(gpCam.getMediaInfo('file'))``` in a script works? Might be because the camera is not connected/turned off/supported