KonradIT/gopro-py-api #10: Media download says Not supported while recording or processing media.

sankhakarfa, 2017-07-19:
I am using gopro Hero 4 silver.
Everytime I connect and capture it works. But when I try downloadLastMedia() option or downloadMedia() it shows "Not supported while recording or processing media."

The self.isRecording() functions returns 1.
What mode should I keep my camera to download the media files?

KonradIT, 2017-07-19:
should not be recording or taking a timelapse, camera should be idle

sankhakarfa, 2017-07-19:
I kept in Playback mode and still it shows error.
Also when I keep it playback mode and use gpCam.overview() the code on line 976 throws error.
Because the mode returns 4.

KonradIT, 2017-07-19:
Put it in video mode or any other mode

sankhakarfa, 2017-07-20:
In Video mode also its not working.
output to the commands:

```
HERO4 Silver
HD4.01.05.00.00
Camera successfully connected!
Not supported while recording or processing media.
```

I checked http://10.5.5.9/gp/gpControl and got the key pair for status / system
8 corresponds to systembusy and in your code you are using it to check if it is recording or not.
Hope this helps.
```
"group": "system",
"fields": [
{
"id": 1,
"name": "internal_battery_present"
},
{
"id": 2,
"name": "internal_battery_level"
},
{
"id": 3,
"name": "external_battery_present"
},
{
"id": 4,
"name": "external_battery_level"
},
{
"id": 6,
"name": "system_hot"
},
{
"id": 8,
"name": "system_busy"
},
{
"id": 9,
"name": "quick_capture_active"
},
{
"id": 10,
"name": "encoding_active"
},
{
"id": 11,
"name": "lcd_lock_active"
},
{
"id": 45,
"name": "camera_locate_active"
},
{
"id": 57,
"name": "current_time_msec"
},
{
"id": 60,
"name": "next_poll_msec"
},
{
"id": 61,
"name": "analytics_ready"
},
{
"id": 62,
"name": "analytics_size"
},
{
"id": 63,
"name": "in_contextual_menu"
}
]
},
```

KonradIT, 2017-07-20:
get this url: http://10.5.5.9/gp/gpControl/status and post the output here.

sankhakarfa, 2017-07-20:
Here is the output.

```
{
"status": {
"1": 1,
"2": 3,
"3": 0,
"4": 0,
"6": 0,
"8": 1,
"9": 0,
"10": 0,
"11": 0,
"13": 0,
"14": 0,
"15": 0,
"16": 0,
"17": 1,
"19": 1,
"20": 1,
"21": 0,
"22": 0,
"23": 0,
"24": 0,
"26": 0,
"27": 0,
"28": 64,
"29": "",
"30": "GP27975140",
"31": 0,
"32": 0,
"33": 0,
"34": 0,
"35": 3759,
"36": 19,
"37": 3,
"38": 282,
"39": 3,
"40": "%11%07%14%14%00%3A",
"41": 0,
"42": 0,
"43": 5,
"44": 0,
"45": 0,
"46": 1,
"47": 1,
"48": 1,
"49": 0,
"54": 14925088,
"55": 0,
"56": 4,
"57": 38807,
"58": 0,
"59": 0,
"60": 500,
"61": 2,
"62": 0,
"63": 0,
"64": 0,
"65": 0,
"66": 0,
"67": 0
},
"settings": {
"1": 0,
"2": 9,
"3": 5,
"4": 0,
"5": 0,
"6": 1,
"7": 1,
"8": 1,
"9": 0,
"10": 0,
"11": 0,
"12": 0,
"13": 1,
"14": 0,
"15": 4,
"16": 0,
"17": 0,
"18": 0,
"19": 0,
"20": 0,
"21": 0,
"22": 0,
"23": 0,
"24": 0,
"25": 0,
"26": 4,
"27": 1,
"28": 0,
"29": 5,
"30": 1,
"31": 0,
"32": 0,
"33": 0,
"34": 0,
"35": 0,
"36": 0,
"37": 0,
"38": 0,
"39": 4,
"49": 0,
"50": 1,
"51": 1,
"52": 1,
"53": 2,
"54": 0,
"55": 2,
"56": 0,
"57": 0,
"58": 1,
"59": 0,
"60": 8,
"61": 1,
"62": 700000,
"63": 0,
"64": 1,
"68": 0,
"69": 0,
"70": 1,
"72": 1,
"73": 0,
"74": 0,
"75": 3,
"76": 3,
"77": 0,
"78": 0,
"79": 0,
"80": 1,
"81": 0,
"82": 0
}
}
```

KonradIT, 2017-07-20:
>"status"

>"8": 1

This number needs to be 0. I lost my H4 so I can't test but it only is 1 when is recording/taking a timelapse/etc... modify the python if you want so it does not check but I doubt it will let you download any video from the camera.

sankhakarfa, 2017-07-21:
Thanks. Changing the code I am able to download photos.

sankhakarfa, 2017-07-26:
Just and Update. Earlier I was using Wifi AP mode on the HERO 4 to connect. 
After making my own access point and connecting my HERO 4's with that it works. 
Status is zero.
Can you update your readme with this link 
http://goprohero.readthedocs.io/en/latest/Wifi%20Research/
This helped me to connect multiple cameras and control with your code.

closed 2017-07-26 by sankhakarfa

KonradIT, 2017-07-26:
you connected multiple cameras to what?

sankhakarfa, 2017-07-26:
I am using a raspberry pi , it acts like a gopro Remote.
But basically any wifi AP can be used to connect multiple cameras.

KonradIT, 2017-07-26:
ahhh got it, and it works with my API? I thought it uses a different protocol when using remote

sankhakarfa, 2017-07-26:
I can use this commands https://github.com/KonradIT/goprowifihack/blob/master/HERO4/WifiCommands.md
But the ip is different assigned by me. 
So I have a DHCP server which assigns static IP for each GO Pro Mac.
And then using these commands I am able to control it.

Basically the Go Pro Remote broadcasts AP at 

> HERO-RC-XXXXX
Create a hostspot and dhcp server with this AP and then the GoPros will connect with it when wifi remote is selected on the camera.
Then everything is almost same except the IP and Downloading Media is also possible.

KonradIT, 2018-03-10:
@selimonat is the camera recording of processing? Might try: 

```
gpCam.take_photo(4)
time.sleep(5)
gpCam.downloadLastMedia()
```

selimonat, 2018-03-10:
no help. The photo is taken but no download. Also I am not sure what does the "ÿûAllow" want to tell me, doesnt look that healthy.

```
In [1]: from goprocam import constants

In [2]: from goprocam import GoProCamera

In [3]: gpCam = GoProCamera.GoPro()
HTTP/1.1 ÿûAllow: GET 

None

In [4]: gpCam.take_photo(4)
bx00x02nx0cHD3100302x15HERO3SilverEdition
bx00x02nx0cHD3100302x15HERO3SilverEdition
wait 4 seconds.
Out[4]: ''

In [5]: gpCam.downloadLastMedia()
Not supported while recording or processing media.

In [6]: 
```

I am on

```
pi@homebrain:~/code/python/gopro-py-api $ python --version
Python 3.4.3 :: Continuum Analytics, Inc.
```

selimonat, 2018-03-10:
to be on the safe side, I upgraded my python to 3.6 as you mention it. Already without taking any photos or doing anything with the camera, this is what I get:

```
(py36)pi@homebrain:~/code/python/gopro-py-api $ python
Python 3.6.0 | packaged by rpi | (default, Jan  1 2017, 01:11:40) 
[GCC 4.9.2] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from goprocam import GoProCamera
>>> from goprocam import constants
>>> gpCam = GoProCamera.GoPro()
HTTP/1.1 ÿûAllow: GET 

None
gp>>> gpCam.IsRecording(
... )
1
```

KonradIT, 2018-03-10:
@selimonat Thanks. Can you do this:

1) Make sure the camera is in standby mode and is not shooting a timelapse/recording
2) Output of `gpCam.getStatus(constants.Hero3Status.IsRecording)` and `gpCam.getStatusRaw()`

selimonat, 2018-03-10:
funny, this time I got an error when creating GoProCamera instance, but this is probably because the camera was in standby mode. The second call went through... Please see below what you wanted to see, the first one didn't give any input, whereas the second one is not human readable...

```
(py36)pi@homebrain:~/code/python/gopro-py-api $ python
Python 3.6.0 | packaged by rpi | (default, Jan  1 2017, 01:11:40) 
[GCC 4.9.2] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from goprocam import constants;from goprocam import GoProCamera;gpCam = GoProCamera.GoPro();
HTTP/1.1 ÿûAllow: GET 

None
Traceback (most recent call last):
  File "/home/pi/miniconda3/envs/py36/lib/python3.6/http/client.py", line 283, in _read_status
    status = int(status)
ValueError: invalid literal for int() with base 10: 'ÿûAllow:'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/code/python/gopro-py-api/goprocam/GoProCamera.py", line 144, in whichCam
    response_raw = urllib.request.urlopen('http://' + self.ip_addr + '/gp/gpControl', timeout=5).read().decode('utf8')
  File "/home/pi/miniconda3/envs/py36/lib/python3.6/urllib/request.py", line 223, in urlopen
    return opener.open(url, data, timeout)
  File "/home/pi/miniconda3/envs/py36/lib/python3.6/urllib/request.py", line 526, in open
    response = self._open(req, data)
  File "/home/pi/miniconda3/envs/py36/lib/python3.6/urllib/request.py", line 544, in _open
    '_open', req)
  File "/home/pi/miniconda3/envs/py36/lib/python3.6/urllib/request.py", line 504, in _call_chain
    result = func(*args)
  File "/home/pi/miniconda3/envs/py36/lib/python3.6/urllib/request.py", line 1346, in http_open
    return self.do_open(http.client.HTTPConnection, req)
  File "/home/pi/miniconda3/envs/py36/lib/python3.6/urllib/request.py", line 1321, in do_open
    r = h.getresponse()
  File "/home/pi/miniconda3/envs/py36/lib/python3.6/http/client.py", line 1331, in getresponse
    response.begin()
  File "/home/pi/miniconda3/envs/py36/lib/python3.6/http/client.py", line 297, in begin
    version, status, reason = self._read_status()
  File "/home/pi/miniconda3/envs/py36/lib/python3.6/http/client.py", line 287, in _read_status
    raise BadStatusLine(line)
http.client.BadStatusLine: HTTP/1.1 ÿûAllow: GET 

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/code/python/gopro-py-api/goprocam/GoProCamera.py", line 50, in __init__
    self._camera = self.whichCam()
  File "/home/pi/code/python/gopro-py-api/goprocam/GoProCamera.py", line 181, in whichCam
    response = urllib.request.urlopen('http://' + self.ip_addr + '/camera/cv',timeout=5).read()
  File "/home/pi/miniconda3/envs/py36/lib/python3.6/urllib/request.py", line 223, in urlopen
    return opener.open(url, data, timeout)
  File "/home/pi/miniconda3/envs/py36/lib/python3.6/urllib/request.py", line 532, in open
    response = meth(req, response)
  File "/home/pi/miniconda3/envs/py36/lib/python3.6/urllib/request.py", line 642, in http_response
    'http', request, response, code, msg, hdrs)
  File "/home/pi/miniconda3/envs/py36/lib/python3.6/urllib/request.py", line 570, in error
    return self._call_chain(*args)
  File "/home/pi/miniconda3/envs/py36/lib/python3.6/urllib/request.py", line 504, in _call_chain
    result = func(*args)
  File "/home/pi/miniconda3/envs/py36/lib/python3.6/urllib/request.py", line 650, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
urllib.error.HTTPError: HTTP Error 410: Gone

>>> from goprocam import constants;from goprocam import GoProCamera;gpCam = GoProCamera.GoPro();
HTTP/1.1 ÿûAllow: GET 

None
>>> gpCam.getStatus(constants.Hero3Status.IsRecording)
>>> gpCam.getStatusRaw()
b'\x00\x00\x00\x00\x01\x00\x00\x01\x08,\x00\xff\x00\x00\x00\x00\x00\x02\x90\x03\x00\x14\x88\x04\x85\x00q\x00~\x00\x04\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x03\x04\x00\x00\x00\x00'
```

zongoalbert, 2019-11-25:
hello , i have the same problem with my hero 5 black , i can't download the media : it's give my this message : Not supported while recording or processing media , please help

zongoalbert, 2019-11-25:
> Thanks. Changing the code I am able to download photos.

Hello SIr , how you fixed this problem.

sankhakarfa, 2019-11-26:
> > Thanks. Changing the code I am able to download photos.
> 
> Hello SIr , how you fixed this problem.

Hey there is a check on line number 791 in GoProCamera.py 
Just remove that condition and check.

Pounii25, 2020-11-21:
Hi all,

I have same error, unable to dowload media from my hero 7 black. I tried in all mode but still receive "Not supported while recording or processing media." Already checked the status page like asked before, and the 8 is set to 0. I note a strange value in 4, which is 255. Any idea? Thank you!!

Pounii25, 2020-11-21:
I just tried to downgrade to 4.0.3 and working fine! :)

KonradIT, 2020-11-22:
Weird, so the value 8 is set to 0, and it said Not supported? I'll check the diff between 4.0.3 and the latest and try to fix it, i use this package nearly every day to download media via USB on the 9 and  it works.

KonradIT, 2020-12-08:
Hey guys, quick update:

I did not encounter my camera replying 1 to the 8 key, which is status, rather I encountered a bug where `ready` was checked against `"1"` instead of int `1` 🤦 . Fixed and pushed. Now it works with my Hero5 Black which is running the **latest** firmware from gopro.com

````
>>> from goprocam import GoProCamera, constants
>>> gopro = GoProCamera.GoPro()
HERO5 Black
HD5.02.02.70.00
Camera successfully connected!
>>> gopro.take_photo()
'http://10.5.5.9/videos/DCIM/101GOPRO/GOPR4391.JPG'
>>> gopro.downloadLastMedia()
filename: GOPR4391.JPG
size: 2.76MB
>>>
````

sahajamatya, 2020-12-23:
I got the same error with my Fusion, takes a picture but fails to download. 

Here's my code: 

```
from goprocam import GoProCamera, constants                                 

gpCam = GoProCamera.GoPro()
gpCam.downloadLastMedia(gpCam.take_photo(1))
```

bash output: 
$ python takepic.py 
```
FUSION
FS1.04.01.80.00
Camera successfully connected!
Not supported while recording or processing media.
```
 
Status output from http://10.5.5.9/gp/gpControl/status:
```
1 | 1
2 | 1
3 | 0
4 | 0
6 | 0
8 | 0
9 | 0
10 | 0
13 | 0
17 | 1
19 | 0
20 | 0
21 | 0
22 | 0
23 | 0
24 | 0
26 | 0
27 | 0
28 | 82
29 | ""
30 | "SahajAmatya"
31 | 0
32 | 0
33 | 0
34 | 5368
35 | 4018
36 | 25
37 | 13
38 | 25
39 | 13
40 | "%14%0C%17%0B%0B%1E"
41 | 0
42 | 0
43 | 1
44 | 0
45 | 0
46 | 1
47 | 1
48 | 1
49 | 0
54 | 27724112
55 | 1
56 | 4
57 | 786511
58 | 0
59 | 0
60 | 500
61 | 2
62 | 0
63 | 0
64 | 0
65 | 0
66 | 0
67 | 0
68 | 1
69 | 1
70 | 27
71 | 12
72 | 16
73 | 20
74 | 0
76 | 1
80 | 0
81 | 1
82 | 1
settings |  
1 | 0
2 | 14
3 | 8
5 | 0
10 | 0
13 | 1
15 | 4
16 | 0
17 | 12
19 | 1
21 | 0
24 | 0
26 | 4
27 | 0
28 | 12
29 | 5
30 | 0
31 | 1
32 | 3601
34 | 0
37 | 0
39 | 4
54 | 1
57 | 0
60 | 8
61 | 1
62 | 2500000
64 | 10
68 | 0
69 | 0
70 | 1
74 | 0
82 | 0
83 | 1
84 | 0
85 | 0
86 | 0
87 | 40
88 | 100
91 | 2
92 | 16
95 | 0

```