KonradIT/gopro-py-api #11: Launch ffplay with python - Android

5perseo, 2017-08-12:
1) In order to have one self contained py script to go in streaming with session 4 which command line in python of the equivalent  ffplay -an -fflags nobuffer -f:v mpegts -probesize 8192 rtp://10.5.5.9:8554?

2)possibility to do the same by android. API could work on SMARTPHONE? 
Used GoProStream.py on samsung S5 and works properly but none ffmpeg app is able to stream on udp 8554 ..

KonradIT, 2017-08-12:
1) what?

2) on Android you can stream to udp://127.0.0.1:10000 and then use VLC to play that URL back.

5perseo, 2017-08-12:
1) Searching for a commad like "ffplay -an -fflags nobuffer -f:v mpegts -probesize 8192 rtp://10.5.5.9:8554" (working on pc in dos shell) in order to put it inside GoProStream.py and have only one script ready to stream.
2) I'll try as soon as possible!
Till now: i've done: **On PC** using PY script find here : Wake up (using API for inizialitazion), initializing to stream with GoProStream.py and finally streaming launching ffplay with a .bat (ffplay udp://10.5.5.9:8554 or the command write above on rtp); Unable to stream by VLC on udp://10.5.5.9:8554 ort rtp://10.5.5.9:8554)
 
**On Android** Installed Qpython3.apk and Load goprocam pypi in order to use API for every type of command. Able to run on smarthphone same command like on pc. (wake up cam, ready to stream and finally stream with an ffplayer app on udp://10.5.5.9:8554 (but without possibility to send command in order to reduce delay, etc ,etc, )). VLC app not working (??)

Installed an usb repetear lowcost (WF8300) configured (10.5.5.10) and linked to wifi session 4. On pc connected to the wifi of repetear making same procedure (.py script, without changing nothing ...??) succeed in streaming at a triple distance (!!). Would repeat same job by the smarthphone and trying different repeater .

5perseo, 2017-08-13:
GoProStream.py has the piece of script i need inside actually in order to launch ffmpeg (i change to ffplay -an -fflags nobuffer -f:v mpegts -probesize 8192 rtp://10.5.5.9:8554 that is the command that works well for me too) but streaming won't start automatically and i need to put that command line in dos shell (?)

5perseo, 2017-08-13:
Ok change in GoProStream the "udp://:8554" to "rtp://10.5.5.9:8554" and works pefectly.
Put 
from goprocam import GoProCamera
from goprocam import constants
gpCam = GoProCamera.GoPro(constants.gpcontrol)
in order to wake up...
Now one script goes in streaming.... 'll try pass on Android now. Capture official doesn't accept the repetear it falls the wifi extended down and search the main gopro's ones which i need to ignore in order to get further and underwater for 100 meters..

[image]

5perseo, 2017-08-13:
On Android this error,,,, maybe have to find a library for Qpython...

[image]

KonradIT, 2017-08-13:
Use Termux on Android and install ffmpeg there.

KonradIT, 2017-08-13:
and use VLC, ffplay does not work with Android

5perseo, 2017-08-14:
It s a pity there s no way to launch ffplay from py script... i hope to
have one script for everything and not  jump from a platform to
another..besides  opening mmplayer app i get streaming on udp port but i
was hoping in some qpython3 library...

Vlc won't work for me nor on PC neither on android ... i change http to rtp
protocol as the reccomand but 😭
1) For android should i change streaming chanel as You told me from
10.5.5.9:8554 to  127.0.0.1:5000 ? Have i to do this in your GoProStream.py
before try to get streaming over Vlc??

2) Dowload tremux and installed mmpeg library .. don't know Linux Commad for
"ffplay  -loglevel panic -fflags nobuffer -f:v mpegts -probesize 8192 rtp://
10.5.5.9:8554" to get streaming..
Could You Tel me please?

Thanks in advance..
Paolo

5perseo, 2017-08-23:
termux not find library able to launch ffplay.
With API streaming on 127.0.0.1 :10000 get streaming in VLC on pc but not on android.
But 5 sec of latency.... is there a command like ffplay to reduce latency?
Changing port i'm unable to come back to port 10.5.5.9:8554 (also launchng GoProStream.py original with port 8554) unless I restart PC (???).