KonradIT/gopro-py-api #12: Camera hangs when Switching between Livestreaming and Photo mode

Prasanth90, 2017-09-18:
I am currently Hero 4 Session and my use-case is to stream the live feed and when the user clicks a button, I want to take a photo and resume the live feed.

I am able to stream the live feed gopro.stream("udp://127.0.0.1:1000", "high") and capturing the feed in the browser

I also wanted to take picture in the mid of the live streaming. I tried

1)  gopro.take_photo(), 

2)  gopro.livestream("stop")
     gopro.take_photo()
     gopro.stream("udp://127.0.0.1:1000", "high")

But the camera hangs in both the cases, the ISBUSY flag is set indefinitely. (take_photo method doesn't return)

I am calling gopro.stream(.......) in a separate thread.

**Environment:**

- Windows 7 , 64-bit, Python 3.6
- Using the published python package

Am I missing anything ? 

Do I need to call any other function before calling the take_photo method ?

Any clue would be much helpfull

Thanks in Advance

KonradIT, 2017-09-18:
[Just did it with my HERO5 Black](https://streamable.com/gc57i)

Don't have the session working at the moment but it works here.

Prasanth90, 2017-09-18:
Thanks for replying

Please post here if you get sometime to make it work with **Hero Session** or  **HERO 4 Session**

Just wanted to make sure whether I am doing it the right way

I called the following code in a separate thread since its a blocking call.
`gopro.stream("udp://127.0.0.1:1000", "high")`

and I am calling `gopro.take_photo()`  from the main thread.

Regards
Prasanth

Prasanth90, 2017-09-19:
**Update:**

I am able to capture 4 to 5 images when the camera(HERO 4 Session) is in live streaming mode. After that it hangs.

The **ISBUSY** flag is set indefinitely.

Is there anyway where i can forcefully **reset** the **ISBUSY** flag ?

Please advice.

Regards
Prasanth

KonradIT, 2017-09-19:
Not really, if the camera is streaming you can take pictures in another terminal, see my video.

danigna77, 2020-02-05:
I am facing the same problem without streaming. I cant take a couple of pictures. Then the API freezes. How did you solve this @Prasanth90 ?