KonradIT/gopro-py-api #19: failed to receive frame on gopro hero 5

mrpiccolo, 2017-11-07:
hi 
i m triyng to get the live streaming using the example gopro_keepalive.py and gopro_opencv.py but i acquirate a frame, it is empty.

When i launch:
`python3 gopro_keepalive.py  `

i receive:

> HERO5 Black
HD5.02.02.00.00
Camera successfully connected!
{"status":"0" }

ffmpeg version 2.8.11-0ubuntu0.16.04.1 Copyright (c) 2000-2017 the FFmpeg developers
  built with gcc 5.4.0 (Ubuntu 5.4.0-6ubuntu1~16.04.4) 20160609
  configuration: --prefix=/usr --extra-version=0ubuntu0.16.04.1 --build-suffix=-ffmpeg --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --cc=cc --cxx=g++ --enable-gpl --enable-shared --disable-stripping --disable-decoder=libopenjpeg --disable-decoder=libschroedinger --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmodplug --enable-libmp3lame --enable-libopenjpeg --enable-libopus --enable-libpulse --enable-librtmp --enable-libschroedinger --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxvid --enable-libzvbi --enable-openal --enable-opengl --enable-x11grab --enable-libdc1394 --enable-libiec61883 --enable-libzmq --enable-frei0r --enable-libx264 --enable-libopencv
  libavutil      54. 31.100 / 54. 31.100
  libavcodec     56. 60.100 / 56. 60.100
  libavformat    56. 40.101 / 56. 40.101
  libavdevice    56.  4.100 / 56.  4.100
  libavfilter     5. 40.101 /  5. 40.101
  libavresample   2.  1.  0 /  2.  1.  0
  libswscale      3.  1.101 /  3.  1.101
  libswresample   1.  2.101 /  1.  2.101
  libpostproc    53.  3.100 / 53.  3.100
[mpegts @ 0x1f27140] Could not find codec parameters for stream 1 (Audio: aac ([15][0][0][0] / 0x000F), 0 channels, fltp): unspecified sample rate                                                                                                        
Consider increasing the value for the 'analyzeduration' and 'probesize' options                                              
[mpegts @ 0x1f27140] Could not find codec parameters for stream 2 (Unknown: none ([128][0][0][0] / 0x0080)): unknown codec   
Consider increasing the value for the 'analyzeduration' and 'probesize' options                                              
Input #0, mpegts, from 'udp://10.5.5.9:8554':
  Duration: N/A, start: 0.000000, bitrate: N/A
  Program 1 
    Stream #0:0[0x1011]: Video: h264 (Main) ([27][0][0][0] / 0x001B), yuvj420p(pc, bt709), 960x720 [SAR 1:1 DAR 4:3], 25 fps, 25 tbr, 90k tbn, 50 tbc
    Stream #0:1[0x1100]: Audio: aac ([15][0][0][0] / 0x000F), 0 channels, fltp
    Stream #0:2[0x200]: Unknown: none ([128][0][0][0] / 0x0080)
Please use -b:a or -b:v, -b is ambiguous
[swscaler @ 0x21170e0] deprecated pixel format used, make sure you did set range correctly                                   
Output #0, mpegts, to 'udp://127.0.0.1:10000':
  Metadata:
    encoder         : Lavf56.40.101
    Stream #0:0: Video: mpeg2video, yuv420p, 960x720 [SAR 1:1 DAR 4:3], q=2-31, 800 kb/s, 30 fps, 90k tbn, 30 tbc
    Metadata:
      encoder         : Lavc56.60.100 mpeg2video
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (native) -> mpeg2video (native))
Press [q] to stop, [?] for help
[h264 @ 0x200f200] left block unavailable for requested intra mode at 0 5
[h264 @ 0x200f200] error while decoding MB 0 5, bytestream 8470
[h264 @ 0x200f200] concealing 2449 DC, 2449 AC, 2449 MV errors in P frame
[h264 @ 0x2288800] concealing 1865 DC, 1865 AC, 1865 MV errors in P frame
[h264 @ 0x200f200] left block unavailable for requested intra mode at 0 24
[h264 @ 0x200f200] error while decoding MB 0 24, bytestream 3947
[h264 @ 0x200f200] concealing 1309 DC, 1309 AC, 1309 MV errors in P frame
[h264 @ 0x22cff20] concealing 1844 DC, 1844 AC, 1844 MV errors in P frame4kbits/s    
[h264 @ 0x2067d40] concealing 1297 DC, 1297 AC, 1297 MV errors in P frame
[h264 @ 0x2317620] error while decoding MB 46 40, bytestream -6
[h264 @ 0x2317620] concealing 303 DC, 303 AC, 303 MV errors in P frame
[h264 @ 0x2288800] concealing 647 DC, 647 AC, 647 MV errors in P frame
[h264 @ 0x200f200] concealing 678 DC, 678 AC, 678 MV errors in P frame
[h264 @ 0x2317620] concealing 2207 DC, 2207 AC, 2207 MV errors in P frame2kbits/s    
[h264 @ 0x22cff20] error while decoding MB 52 42, bytestream -5rate=1059.1kbits/s    
[h264 @ 0x22cff20] concealing 177 DC, 177 AC, 177 MV errors in P frame
[h264 @ 0x200f200] concealing 1351 DC, 1351 AC, 1351 MV errors in P frame
[h264 @ 0x22cff20] left block unavailable for requested intra mode at 0 8
[h264 @ 0x22cff20] error while decoding MB 0 8, bytestream 7447
[h264 @ 0x22cff20] concealing 2269 DC, 2269 AC, 2269 MV errors in P frame
[h264 @ 0x2067d40] concealing 1535 DC, 1535 AC, 1535 MV errors in P frame
[h264 @ 0x2067d40] concealing 1838 DC, 1838 AC, 1838 MV errors in P frame0kbits/s    
[h264 @ 0x2317620] concealing 2140 DC, 2140 AC, 2140 MV errors in P frame
[h264 @ 0x22cff20] error while decoding MB 13 44, bytestream -5rate= 986.1kbits/s    
[h264 @ 0x22cff20] concealing 96 DC, 96 AC, 96 MV errors in P frame
[h264 @ 0x2067d40] left block unavailable for requested intra mode at 0 36
[h264 @ 0x2067d40] error while decoding MB 0 36, bytestream 1522

what i wrong?

thank you

KonradIT, 2017-11-07:
Hi, that sends the stream to localhost, on another terminal run `python gopro_opencv.py`
[image]

mrpiccolo, 2017-11-08:
thank you for the reply
that is a screen of the error
[image]
i m using ubuntu16.04 and python3 to launch the software

KonradIT, 2017-11-08:
something wrong with your OpenCV installation, Google the error.

mrpiccolo, 2017-11-10:
Thank you for your reply but the problem isn't opencv, because when i try to capture the frame, it is empty,  when i launch gopro_keepalive.py, i receive errors you don't have in your screen

KonradIT, 2017-11-10:
In the first terminal everything is in order, problem is [opencv](https://stackoverflow.com/questions/30506126/open-cv-error-215-scn-3-scn-4-in-function-cvtcolor)