KonradIT/gopro-py-api #20: gpCam.overview()  TypeError: Can't convert 'NoneType' object to str implicitly

Berryfier, 2017-11-26:
Thank you for sharing this API
I'm using goprocam (1.9.3) on python 3 with gopro hero 4 session.
Unfortunately my terminal looks like this:
>>> gpCam.overview()
camera overview
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/.local/lib/python3.4/site-packages/goprocam/GoProCamera.py", line 997, in overview
    print("current mode: " + "" + self.parse_value("mode", self.getStatus(constants.Status.Status, constants.Status.STATUS.Mode)))
TypeError: Can't convert 'NoneType' object to str implicitly

KonradIT, 2017-11-26:
update to 1.9.6

Berryfier, 2017-11-27:
Thank you for your suggestion. I have successfully updated to 1.9.6.
Unfortunately my terminal now looks like this:
>>> gpCam.overview()
camera overview
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/.local/lib/python3.4/site-packages/goprocam/GoProCamera.py", line 995, in overview
    print("current mode: " + "" + self.parse_value("mode", self.getStatus(constants.Status.Status, constants.Status.STATUS.Mode)))
TypeError: Can't convert 'NoneType' object to str implicitly
>>>

KonradIT, 2017-11-27:
what's your full code? do you have the camera connected?

Berryfier, 2017-11-27:
Thank you for giving your time to this. The camera is connected by WiFi (but it has not been paired).  My whole terminal dialogue is:
pi@noobs190:~ $ python3
Python 3.4.2 (default, Oct 19 2014, 13:31:11) 
[GCC 4.9.1] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from goprocam import GoProCamera
>>> from goprocam import constants
>>> gpCam = GoProCamera.GoPro(constants.gpcontrol)
Waking up...
Camera successfully connected!
Connected to 10.5.5.9
>>> gpCam.overview()
camera overview
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/.local/lib/python3.4/site-packages/goprocam/GoProCamera.py", line 995, in overview
    print("current mode: " + "" + self.parse_value("mode", self.getStatus(constants.Status.Status, constants.Status.STATUS.Mode)))
TypeError: Can't convert 'NoneType' object to str implicitly
>>>

KonradIT, 2017-11-27:
you need to pair it first

Berryfier, 2017-11-28:
Thank you KnoradIT. I'll close this whilst I do the python update to 3.6.X needed for gpCam.pair() and re-open if the problem persists.

closed 2017-11-28 by Berryfier