KonradIT/gopro-py-api #21: gpCam.pair AttributeError: 'module' object has no attribute '_create_unverified_context'

Berryfier, 2017-11-27:
Thank you for sharing this API
I'm using goprocam (1.9.6) on python 3 with gopro hero 4 session.
Unfortunately my terminal looks like this:
pi@noobs190:~ $ python3
Python 3.4.2 (default, Oct 19 2014, 13:31:11) 
[GCC 4.9.1] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from goprocam import GoProCamera
>>> from goprocam import constants
>>> gpCam = GoProCamera.GoPro(constants.gpcontrol)
Waking up...
Camera successfully connected!
Connected to 10.5.5.9
>>> gpCam.pair()
Make sure your GoPro camera is in pairing mode!
Go to settings > Wifi > PAIR > GoProApp to start pairing.
Then connect to it, the ssid name should be GOPRO-XXXX/GPXXXXX/GOPRO-BP-XXXX and the password is goprohero
Enter pairing code: 4680
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/.local/lib/python3.4/site-packages/goprocam/GoProCamera.py", line 362, in pair
    context = ssl._create_unverified_context()
AttributeError: 'module' object has no attribute '_create_unverified_context'
>>>

KonradIT, 2017-11-27:
update your python3 to 3.6.* (recommended target for this API)

Berryfier, 2017-11-28:
Thank you KnoradIT. I'll close this whilst I do the python update and re-open if the problem persists.

closed 2017-11-28 by Berryfier

Berryfier, 2018-03-03:
Still awaiting the python update for the pi, but I discovered the following workaround for pair.
In file GoProCamera.py at
line 367. add _ to make _context
lines 369 and 371. add _ to make context=_context

KonradIT, 2018-03-03:
So like this?

	_context = ssl._create_unverified_context()
	ssl._create_default_https_context = ssl._create_unverified_context
	response_raw = urllib.request.urlopen('https://' + self.ip_addr + '/gpPair?c=start&pin=' + code + '&mode=0', context=_context).read().decode('utf8')
	print(response_raw)
	response_raw = urllib.request.urlopen('https://' + self.ip_addr + '/gpPair?c=finish&pin=' + code + '&mode=0', context=_context).read().decode('utf8')

Berryfier, 2018-03-03:
Exactly!
My GoPro is paired and all is working perfectly now (Pi is currently at Python 3.5.3).
Thanks again for making your code available.

KonradIT, 2018-03-08:
Cool this will be added in the next release.