KonradIT/gopro-py-api #23: delete in gopro3+ retuerns error

lavitan, 2017-12-08:
>>> gpCam.delete("last")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/site-packages/goprocam/GoProCamera.py", line 303, in delete
    print(self.sendCamera("DL"))
  File "/usr/local/lib/python3.6/site-packages/goprocam/GoProCamera.py", line 113, in sendCamera
    urllib.request.urlopen('http://10.5.5.9/camera/' + param + '?t=' + self.getPassword() + value_no      tempty, timeout=5).read()
UnboundLocalError: local variable 'value_notempty' referenced before assignment

KonradIT, 2017-12-08:
can you try running the API from source?

lavitan, 2017-12-08:
same response

z3r63ru5, 2017-12-08:
I have the same error (pip3 install goprocam) 
The problem is a typo in line 108 ("value_notem**tp**y") ...

Regards,
DKK

lavitan, 2017-12-10:
thank you, it works