KonradIT/gopro-py-api #26: Gopro Fusion compatibility

hbardak, 2017-12-26:
Hi Konrad, 

Thanks for this lib. I was wondering   if there is any plan to support the new gopro fusion. 
So far from my test you can connect and take_photo(). 
I have found most of the settings of the camera are still compatible such as Protune, EV change, Raw, photo/video mode etc... Only new settings such as resolution, ISO got new parameters that are not listed in the library. 

However since the fusion is a dual camera I can get or download only the back camera image. l'url return only the back camera file. 

Thanks 

H

KonradIT, 2017-12-26:
Hello!

I don't have a Fusion right now, probably will order the next version since I want a smaller form factor (like the 360one). Fusion seems like a really good camera. I actually started working on Fusion API [in goprowifihack](https://github.com/KonradIT/goprowifihack/blob/master/Fusion1/gpControl-Fusion1.json). So far I have the gpControl extracted from the firmware. Seems like Preview should be the same as in GoPro HERO cameras as well.

Regarding the back image or front image, its because the API gets the latest folder, and it downloads the image from there. [Look at this example of 10.5.5.9/gp/gpMediaList from my Hero5](https://gist.github.com/KonradIT/18d9126f9cd81448175f7a3e7c425376), the API will download 104GOPRO/GOPR2366.JPG, because 104GOPRO is the latest folder and GOPR2366.JPG is the latest file.

Fusion does it differently, according to GoPro's website [the camera has 2 folders on /DCIM:](https://gopro.com/help/articles/how_to/how-to-render-videos-in-fusion-studio) 100GBACK and 100GFRNT. So the GBACK is recorded later and its picked up by the API.

I am glad you asked for support for Fusion, in order to get the important data please turn on WiFi, then on your computer or phone's browser go to:

1) http://10.5.5.9/gp/gpControl
2) http://10.5.5.9/gp/gpMediaList
3) http://10.5.5.9/gp/gpControl/status

Post the contents of each of the URLs to this issue. Also, if you can, test if streaming works by doing with the Python API:

```
gopro = GoProCamera.GoPro()
gopro.stream("udp://127.0.0.1:10000")
```

and open udp://127.0.0.1:10000 in a media player (eg: VLC, MPV).

KonradIT, 2017-12-27:
I've pushed a new release 1.9.8 to pypi and github-releases. There is a new function called downloadLastSpherical which should download the FRONT image/video and BACK as well.

hbardak, 2017-12-29:
Hi,
sorry for the delay I was away : 

http://10.5.5.9/gp/gpControl give me :  
[gpControl.txt](https://github.com/KonradIT/gopro-py-api/files/1593958/gpControl.txt)

http://10.5.5.9/gp/gpMediaList is printing : 
[gpMedialist.txt](https://github.com/KonradIT/gopro-py-api/files/1593961/gpMedialist.txt)

finally http://10.5.5.9/gp/gpControl/status is printing : 
 [gpControlStatus.txt](https://github.com/KonradIT/gopro-py-api/files/1593965/gpControlStatus.txt)

hbardak, 2017-12-29:
I have tried this 

`from goprocam import GoProCamera
from goprocam import constants

gpCam = GoProCamera.GoPro(constants.gpcontrol)
gpCam.stream("udp://127.0.0.1:10000")`

I got : 
`Waking up...
Camera successfully connected!
Connected to 10.5.5.9
{"status":"0" }`

But when I tried to open udp://127.0.0.1:10000 with VLC but no success. I got nothing.

KonradIT, 2017-12-29:
Thanks. /gpMediaList only returns BACK folder, try getting http://10.5.5.9/gp/gpMediaList2

KonradIT, 2017-12-29:
Re: preview try using FFplay. Also is there any output in the terminal? You can use this script to further check if the preview is the same: http://github.com/sonof8bits/goprostream

KonradIT, 2017-12-30:
@hbardak can you dump these as well? http://10.5.5.9/gp/gpMediaListEx and http://10.5.5.9/gp/gpMediaOP

hbardak, 2017-12-30:
hey Konrad,

more dump as requested

[gpMedialistEx.txt](https://github.com/KonradIT/gopro-py-api/files/1594536/gpMedialistEx.txt)

[gpMediaOP.txt](https://github.com/KonradIT/gopro-py-api/files/1594537/gpMediaOP.txt)

KonradIT, 2017-12-30:
Thanks, try the new function downloadLastSpherical(), should download both GBACK and GFRNT.

hbardak, 2017-12-31:
Hi Konrad, 

I have just tried to use downloadLastSpherical() 

```
    gpCam.take_photo()
    gpCam.downloadLastSpherical() 
```

but I am getting the following error : 

```
gpCam.take_photo()
{}

{}

Out[19]: 'http://10.5.5.9:8080/videos/DCIM/100GBACK/GPBK0109.JPG'

gpCam.downloadLastSpherical()
filename: GPBK0109.JPG
size: 3.4MB
Traceback (most recent call last):

  File "<ipython-input-20-cc02d2c27cc0>", line 1, in <module>
    gpCam.downloadLastSpherical()

  File "C:\Users\harry\Anaconda3\lib\site-packages\goprocam\GoProCamera.py", line 764, in downloadLastSpherical
    urllib.request.urlretrieve(self.getMedia().replace("BACK","FRNT"), self.getMediaInfo("folder").replace("BACK","FRNT")+"-"+self.getMediaInfo("file"))

  File "C:\Users\harry\Anaconda3\lib\urllib\request.py", line 188, in urlretrieve
    with contextlib.closing(urlopen(url, data)) as fp:

  File "C:\Users\harry\Anaconda3\lib\urllib\request.py", line 163, in urlopen
    return opener.open(url, data, timeout)

  File "C:\Users\harry\Anaconda3\lib\urllib\request.py", line 472, in open
    response = meth(req, response)

  File "C:\Users\harry\Anaconda3\lib\urllib\request.py", line 582, in http_response
    'http', request, response, code, msg, hdrs)

  File "C:\Users\harry\Anaconda3\lib\urllib\request.py", line 510, in error
    return self._call_chain(*args)

  File "C:\Users\harry\Anaconda3\lib\urllib\request.py", line 444, in _call_chain
    result = func(*args)

  File "C:\Users\harry\Anaconda3\lib\urllib\request.py", line 590, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)

HTTPError: Not Found
```
what I can see is that my back image is downloaded but the front one isn't I guess because the error 

H.

KonradIT, 2017-12-31:
Fixed in the latest commit, please "git pull" or download the latest zip file from github to test.

hbardak, 2018-01-01:
unfortunately still have the same issue : 

'''
Traceback (most recent call last):

  File "<ipython-input-24-378a6eca1529>", line 1, in <module>
    runfile('C:/Users/harry/OneDrive/Python_Scripts/gopro/gopro_test.py', wdir='C:/Users/harry/OneDrive/Python_Scripts/gopro')

  File "C:\Users\harry\Anaconda3\lib\site-packages\spyder\utils\site\sitecustomize.py", line 866, in runfile
    execfile(filename, namespace)

  File "C:\Users\harry\Anaconda3\lib\site-packages\spyder\utils\site\sitecustomize.py", line 102, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "C:/Users/harry/OneDrive/Python_Scripts/gopro/gopro_test.py", line 76, in <module>
    gpCam.downloadLastSpherical()

  File "C:\Users\harry\Anaconda3\lib\site-packages\goprocam\GoProCamera.py", line 764, in downloadLastSpherical
    urllib.request.urlretrieve(self.getMedia().replace("BACK","FRNT").replace("BK","FR"), self.getMediaInfo("folder").replace("BACK","FRNT")+"-"+self.getMediaInfo("file").replace("BK","FR"))

  File "C:\Users\harry\Anaconda3\lib\urllib\request.py", line 188, in urlretrieve
    with contextlib.closing(urlopen(url, data)) as fp:

  File "C:\Users\harry\Anaconda3\lib\urllib\request.py", line 163, in urlopen
    return opener.open(url, data, timeout)

  File "C:\Users\harry\Anaconda3\lib\urllib\request.py", line 472, in open
    response = meth(req, response)

  File "C:\Users\harry\Anaconda3\lib\urllib\request.py", line 582, in http_response
    'http', request, response, code, msg, hdrs)

  File "C:\Users\harry\Anaconda3\lib\urllib\request.py", line 510, in error
    return self._call_chain(*args)

  File "C:\Users\harry\Anaconda3\lib\urllib\request.py", line 444, in _call_chain
    result = func(*args)

  File "C:\Users\harry\Anaconda3\lib\urllib\request.py", line 590, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)

HTTPError: Not Found
'''

KonradIT, 2018-01-01:
Try with the new commit.

hbardak, 2018-01-01:
I think I got it : 
when you are trying to retrieve the url for the front camera from 

http://10.5.5.9:8080/videos/DCIM/100GBACK/GPBK0128.JPG
to 
http://10.5.5.9:8080/videos/DCIM/100GFRNT/GPFR0128.JPG 

The later one doesn't exist 

When I type the url of the back camera in my browser I got an image but not with the "front" url. 
if I look at : 
http://10.5.5.9:8080/videos/DCIM/ 
I can see only the Back directory
[image]

hbardak, 2018-01-01:
Additionally I have checked on the SD cards 
The folder path is   ...\DCIM\100GFRNT\GPFR0128.JPG
looks like you can't access to front camera folder using url at the moment ( or least we don't know how )

KonradIT, 2018-01-01:
thanks! will fix it now

KonradIT, 2018-01-01:
Fixed with new feedback. Might need a [traffic dump](https://www.quora.com/Is-there-something-like-Wireshark-or-a-terminal-app-for-iOS-to-see-all-IP-adresses-which-communicate-with-my-iPhone-every-IP-of-the-messaging-apps-every-IP-where-apps-are-connected-to-get-ads) from the app.

hbardak, 2018-01-01:
what do mean by a traffic dump ? how can I provide this to you ?

KonradIT, 2018-01-01:
what device are you using to communicate with the fusion? Android or iPhone?

hbardak, 2018-01-04:
I am using a surface pro while using this API. 
Otherwise I got an Iphone. 
IF you can point me on how I can provide you a traffic dump, I will with no hesitation

KonradIT, 2018-01-04:
As you mentioned earlier, [XXX]GFRNT/ is in /DCIM/ so if you could please go to http://10.5.5.9:8080/videos/DCIM/100GBACK/ and pick a media item (say, GPBK0128.JPG as mentioned earlier), make sure you can view it on your browser. then go to http://10.5.5.9:8080/DCIM/100GFRNT and see if there are any media items here. You should be able to see http://10.5.5.9:8080/DCIM/100GFRNT/GPFR0128.JPG (if that's the media item you choose earlier).

hbardak, 2018-01-06:
nope as said before [XXX]GFRNT isn't in DCIM.
[image]
This a screenshot showing that [XXX]GFRNT isn't in DCIM
So I can't access to the front picture using url's.
Back picture are accessible not front one

KonradIT, 2018-01-06:
Is this URL accessible? http://10.5.5.9:8080/DCIM/100GFRNT

hbardak, 2018-01-07:
No unfortunately. 
This is what I am trying to explain. 
At the moment you can't access the FRNT directory from an url.

KonradIT, 2018-01-07:
Ok, to sniff packets from an iphone use [this guide](https://blog.websecurify.com/2017/03/ios-packet-sniffing.html). I don't have an iphone myself so I cannot verify if it works. Then when everything is set up go to the GoPro app, download an image from fusion (which should include back and front) and then look for "10.5.5.9:8080" GET requests in the sniffer. [Paros](https://jerodsanto.net/2009/06/sniff-your-iphones-network-traffic/) might be a good alternative.

santiblanko, 2018-10-10:
Hey check this path http://10.5.5.9/videos2/DCIM/100GFRNT/GPFR0314.JPG

KonradIT, 2018-10-10:
@santiblanko does that work?

santiblanko, 2018-10-10:
Yes sir. I use wireshark to get the request. If you need more fusion commands I can help you. I connect the ipad to the usb and use the gopro fusion app to intercept the http request.

KonradIT, 2018-10-13:
This will be fixed in 3.0.0;

Thanks to @hbardak and @santiblanko for helping!

## How Fusion support works:

- getMedia() = now returns an array with FRONT and BACK media
- listMedia() = list now adjusts to Fusion files
- downloadLastMedia() = untested

Other features untested and not worked on ATM, PRs welcome.

safeisnotanoption, 2018-10-15:
Sadly, this path (http://10.5.5.9/videos2/DCIM/100GFRNT/GPFR0***.JPGorMP4) doesn't work for me.

safeisnotanoption, 2018-10-15:
My bad, looks like sometimes it works, sometimes it doesn't.

KonradIT, 2018-10-15:
Hi @safeisnotanoption , do you know why? If it doesn't work, what's the response?

safeisnotanoption, 2018-10-15:
Don't know why, /gp/gpMediaList still gives me filename of front and back videos,  but /videos2/DCIM/100GFRNT/filename.mp4 gives me "Error 404: Not Found".

KonradIT, 2018-10-15:
Hmm, can you browse 10.5.5.9/videos2 ?

safeisnotanoption, 2018-10-15:
Browsing folders always gives me "Error 404: Not Found"

KonradIT, 2018-10-15:
How about 10.5.5.9/videos2/DCIM/ ?

santiblanko, 2018-10-15:
@safeisnotanoption my skype is @santiblanko I am working with gopro fusion Now and I don't have problems.

safeisnotanoption, 2018-10-16:
@KonradIT No matter what folder, it always gives 404.
@santiblanko Thanks, I think this inconsistency is caused by problems with the hardware (probably my pc wi-fi receiver).

KonradIT, 2018-10-16:
@santiblanko Are the new commits giving you issues? Anything not working/missing?

santiblanko, 2018-10-16:
I don't use the library. I am using the http api. I have a little question, How long is the camera on standby? He visto que vives en España, cuanto tiempo dura la cámara apagada con el wifi activo?

KonradIT, 2018-10-17:
@santiblanko 2-3 horas.

KonradIT, 2018-11-14:
## Fusion support:

Fusion support has been added, thanks to GoPro for providing me with one!

Commit: e51c56b7f6c3f97a4fe62d2945228c9093947908

Fixed:

- `getMedia`
- `listMedia`
- `downloadLastMedia`
- `downloadMedia`
- `overview`
- Added Fusion-specific constants to `constants.py`

Download goprocam-3.0.1 from Pypi or from GitHub releases to try it out.

closed 2019-01-23 by KonradIT