KonradIT/gopro-py-api #28: GoPro Fusion - overview() is erroing

hbardak, 2017-12-29:
Hi Konrad, 

I have just noticed that the overview() method return a error. 

```
gpCam = GoProCamera.GoPro(constants.gpcontrol)
gpCam.overview()
```

return this : 

```
camera overview
current mode: Video
current submode: Video
Traceback (most recent call last):

  File "<ipython-input-9-c0b03d50bfb9>", line 1, in <module>
    gpCam.overview()

  File "C:\Users\harry\Anaconda3\lib\site-packages\goprocam\GoProCamera.py", line 1027, in overview
    print("current video resolution: " + "" + self.parse_value("video_res",self.getStatus(constants.Status.Settings, constants.Video.RESOLUTION)))

TypeError: Can't convert 'NoneType' object to str implicitly
```

KonradIT, 2017-12-30:
Try now with the latest commits

KonradIT, 2018-11-14:
Fixed in e51c56b7f6c3f97a4fe62d2945228c9093947908 . Tested on Fusion 1 with 1.80 FW.

closed 2018-11-14 by KonradIT