KonradIT/gopro-py-api #36: [Hero 6] Pairing screen stays visible, https pairing refuses connection

Calcyfer, 2018-02-21:
I tried pairing the hero 6 through its LCD display (connections -> connect new device -> GoPro App). then connected to the correct SSID and tried .pair(), but the https port refuses connections.

All other commands work, but only while the LCD shows "pairing".

I've also tried configuring a router for the smart remote mode (correct MAC, SSID, DHCP), but in that mode, none of the ports are accessible.

Has anyone had success pairing the Hero6 with the latest firmware?

KonradIT, 2018-03-07:
What's the terminal output when you pair?

KonradIT, 2018-07-13:
For everyone wondering: 

    [gpcam] = GoPro...
    [gpcam].pair(usepin=False) \#For HERO5, 6.

closed 2018-07-13 by KonradIT