KonradIT/gopro-py-api #37: GoPro 5 cannot stream to VLC or OpenCV

kerrias, 2018-02-27:
Hello:

First, thanks for all of your work on this api -- it rocks!  

I am having issues streaming from a GoPro 5 to OpenCV and VLC.  I first run the gopro_keepalive.py script and then I run the gopro_opencv.py script (and get an error that when Googled tells me that OpenCV did not receive a video stream), or VLC (which doesn't display anything, and upon stopping VLC, get an error that according to a Google search signifies that VLC did not receive a video stream).   

I have attached two screen shots that show the keepalive script running in one window, and opencv and/or VLC in the other.  Any insight or advice you have on what I could be doing wrong is greatly appreciated!  

Thanks!
Kerri
[image]
[image]

kerrias, 2018-02-27:
One more piece of information that might be helpful.  Upon killing the keepalive script I get the following:
[image]

KonradIT, 2018-02-27:
In the first picture:

Make sure the left terminal has some FFmpeg output. You need to wait some time for FFmpeg to start transcoding. Also fix that OpenCV error.

**gopro_keepalive.py** should look like this:

[image]

---

**This is gopro_keepalive.py and gopro_opencv.py at the same time**

[image]

--

**This is ffmpegless_preview.py**

[image]

kerrias, 2018-02-27:
Thanks for your quick reply!

The keepalive script has been running for over 60 minutes now and it has not moved beyond the following state.  How long should it take?  

How is your camera set up?  For instance, do you have an SDCard?  Are there specific versions of FW I need to update to?  Any other ideas?

Thanks again!

[image]

KonradIT, 2018-02-27:
Update the firmware and use an up to date version of FFmpeg

kerrias, 2018-03-01:
Awesome!  Updating the FW on my GoPro fixed the issue.  Thank you for your support!  

I have a few questions for you:

1. The delay between me moving the camera and the feed on the computer (OpenCV or ffplay) getting updated is really large (6-10 seconds).  The delay is still quite large (6-ish seconds) when I run the OpenCV script and take out all of the processing that would be looking for faces.  Is this normal?  Could this be due to some WiFi setting that I have on my GoPro that needs changed/modified?  What does your lag or delay look like when you are running the OpenCV example?

2. Do you have experience with reading sensor data off of the video or transport stream that is sent from the GoPro to the computer.  I would like to read GPS, altimeter, and gyro on a frame by frame basis.  Ideas?

Thanks again!

KonradIT, 2018-03-01:
the delay is cumulative in OpenCV, FFplay should be about 1 second or less. for reading sensor data from a video use http://github.com/stilldavid/gopro-utils

pderian, 2018-06-25:
This may be a bit old, anyway... Here's the list of parameters I use with ffplay to start asap and reduce the lag:
 ```ffplay -probesize 32 -sync video -an -sn -fast -infbuf -f mpegts -i udp://...:8554```

DigitalHubRocks, 2018-07-04:
Hi, can I check if the latest firmware HD5.02.02.60.00 is supported for video streaming on Hero5 Black? I am facing the same issues as  @kerrias had at the start. After pairing successfully, running gpcam.stream("udp://127.0.0.1:10000") gives the ffmpeg outputs till the "libpostproc" line and it gets stuck. Is the camera not set up for streaming? Seems like no video stream is coming in. 

I have updated to the latest firmware, and ffmpeg is the latest version (2.8.14-0 ubuntu0.16.04.1) as well. 

Not sure if this is relevant info, but trying to ping 10.5.5.9 returns Destination Port Unreachable.

KonradIT, 2018-07-13:
Thanks @pderian !

closed 2018-07-13 by KonradIT