KonradIT/gopro-py-api #47: GoPro 6 can get informations but cannot stream to VLC.

Neokyuubi, 2018-06-29:
Hello,
Thank you for this amazing API.

I tried it on Windows 7, i can get the camera connected and get the overview infos, but when it came to streaming it fails.
I'm on the firmware 1.60.00, using pythonV3.7.0, and last ffmpegV4 64-bit (i also tried ffmpegV4 32bit just in case), and finally VLC-v3.03 to capture the stream using udp://localhost:5000.

Here's my code for quick testing : 

```
from goprocam import GoProCamera
# from goprocam import constants

gpCam = GoProCamera.GoPro()
# gpCam.pair(usepin=False)  # I use this as said to pair for the first time.

print(gpCam.overview())  # I get the camera infos here but isRecording == Recording even if it's not !!!

gpCam.stream("udp://localhost:5000")  # I tried to switch order with gpCam.livestream("start") just in case but nothing
gpCam.livestream("start")

while True:  # just for quick testing
    print()
```
---------------

And here's the output :

```
ffmpeg version N-91378-g3f953379e1 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 7.3.0 (GCC)
  configuration: --enable-gpl --enable-version3 --enable-sdl2 --enable-bzlib --enable-fontconfig --enable-gnutls --enable-iconv --enable-libass --enable-libbluray --enable-libfreetype --enable-libmp3lame --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopenjpeg --enable-libopus --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libtheora --enable-libtwolame --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxml2 --enable-libzimg --enable-lzma --enable-zlib --enable-gmp --enable-libvidstab --enable-libvorbis --enable-libvo-amrwbenc --enable-libmysofa --enable-libspeex --enable-libxvid --enable-libaom --enable-libmfx --enable-amf --enable-ffnvcodec --enable-cuvid --enable-d3d11va --enable-nvenc --enable-nvdec --enable-dxva2 --enable-avisynth
  libavutil      56. 18.102 / 56. 18.102
  libavcodec     58. 20.104 / 58. 20.104
  libavformat    58. 17.101 / 58. 17.101
  libavdevice    58.  4.101 / 58.  4.101
  libavfilter     7. 25.100 /  7. 25.100
  libswscale      5.  2.100 /  5.  2.100
  libswresample   3.  2.100 /  3.  2.100
  libpostproc    55.  2.100 / 55.  2.100
[h264 @ 00997680] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 00997680] decode_slice_header error
[h264 @ 00997680] no frame!
[h264 @ 00997680] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 00997680] decode_slice_header error
[h264 @ 00997680] no frame!
[h264 @ 00997680] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 00997680] decode_slice_header error
[h264 @ 00997680] no frame!
[h264 @ 00997680] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 00997680] decode_slice_header error
[h264 @ 00997680] no frame!
[h264 @ 00997680] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 00997680] decode_slice_header error
[h264 @ 00997680] no frame!
[h264 @ 00997680] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 00997680] decode_slice_header error
[h264 @ 00997680] no frame!
[h264 @ 00997680] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 00997680] decode_slice_header error
[h264 @ 00997680] no frame!
[h264 @ 00997680] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 00997680] decode_slice_header error
[... 96 lines cut ...]
[h264 @ 00a40000] concealing 1135 DC, 1135 AC, 1135 MV errors in P frame
[h264 @ 00a40000] concealing 1092 DC, 1092 AC, 1092 MV errors in P frame
[h264 @ 00a40000] cabac decode of qscale diff failed at 0 16
[h264 @ 00a40000] error while decoding MB 0 16, bytestream 1526
[h264 @ 00a40000] concealing 791 DC, 791 AC, 791 MV errors in P frame
[h264 @ 05f86c00] cabac decode of qscale diff failed at 6 7
[h264 @ 05f86c00] error while decoding MB 6 7, bytestream 3098
[h264 @ 05f86c00] concealing 1262 DC, 1262 AC, 1262 MV errors in P frame
[h264 @ 05f86c00] concealing 1045 DC, 1045 AC, 1045 MV errors in P frame
```

---------------

And sometimes is starts with : 
```
[mpegts @ 0000000000502d40] PES packet size mismatch
    Last message repeated 1 times
[mpegts @ 0000000000502d40] Could not find codec parameters for stream 2 (Unknown: none ([128][0][0][0] / 0x0080)): unknown codec
Consider increasing the value for the 'analyzeduration' and 'probesize' options
```
Do i need to upgrade to the last firmware 2.01 ?  i hope not, it's a disaster ^^"
Can you help with this one please ?

Thank you so much for keeping this a live.
Karam.

pderian, 2018-06-29:
Hi,

`gpCam.stream()` starts the stream for the GoPro, then spawns ffmepg which receives the stream, possibly re-encodes it and sends it somewhere else (where VLC would read it). From your output it looks like ffmpeg receives the stream from the GoPro.  Maybe the issue is between ffmpeg and VLC?

Since you installed ffmpeg, perhaps ffplay is available as well? If so, you can try to display directly the stream coming from your GoPro:

1.  after connecting to the camera, simply start the stream with `gpCam.livestream("start")`. Do not use `gpCam.stream(...)`.
2. start ffplay with the following options:
```ffplay -f mpegts -i udp://<GOPRO_IP>:8554```
After some startup time, it should display the video on your screen. Use 'q' to quit.

If the above works, then the issue is not with the camera...
Cheers,

Pierre

Neokyuubi, 2018-06-29:
Hi,
Thank you @pderian .

Using ffplay as you said (in my case : `ffplay -f mpegts -i udp://10.5.5.100:8554`), it works for 3 to 4 seconds and then it freeze !

It get stuck at last line of the output : `22.08 M-V: -0.908 fd=   0 aq=    0KB vq=    0KB sq=    0B f=0/0 `, 
But these values keep changing : `22.08 M-V: -0.908`
I waited about 2 minutes but it still frozen !

And here's the output :

```
ffplay version N-91378-g3f953379e1 Copyright (c) 2003-2018 the FFmpeg developers
  built with gcc 7.3.0 (GCC)
  configuration: --enable-gpl --enable-version3 --enable-sdl2 --enable-bzlib --enable-fontconfig --enable-gnutls --enable-iconv --enable-libass --enable-libbluray --enable-libf
reetype --enable-libmp3lame --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopenjpeg --enable-libopus --enable-libshine --enable-libsnappy --enable-libsoxr --
enable-libtheora --enable-libtwolame --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxml2 --enable-libzimg --enable-lzma --en
able-zlib --enable-gmp --enable-libvidstab --enable-libvorbis --enable-libvo-amrwbenc --enable-libmysofa --enable-libspeex --enable-libxvid --enable-libaom --enable-libmfx --en
able-amf --enable-ffnvcodec --enable-cuvid --enable-d3d11va --enable-nvenc --enable-nvdec --enable-dxva2 --enable-avisynth
  libavutil      56. 18.102 / 56. 18.102
  libavcodec     58. 20.104 / 58. 20.104
  libavformat    58. 17.101 / 58. 17.101
  libavdevice    58.  4.101 / 58.  4.101
  libavfilter     7. 25.100 /  7. 25.100
  libswscale      5.  2.100 /  5.  2.100
  libswresample   3.  2.100 /  3.  2.100
  libpostproc    55.  2.100 / 55.  2.100
[h264 @ 00000000003e7440] non-existing PPS 0 referenced  0B f=0/0
    Last message repeated 1 times
[h264 @ 00000000003e7440] decode_slice_header error
[h264 @ 00000000003e7440] no frame!
[h264 @ 00000000003e7440] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 00000000003e7440] decode_slice_header error
[h264 @ 00000000003e7440] no frame!
[h264 @ 00000000003e7440] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 00000000003e7440] decode_slice_header error
[h264 @ 00000000003e7440] no frame!
[h264 @ 00000000003e7440] non-existing PPS 0 referenced  0B f=0/0   
    Last message repeated 1 times
[h264 @ 00000000003e7440] decode_slice_header error
[h264 @ 00000000003e7440] no frame!
[h264 @ 00000000003e7440] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 00000000003e7440] decode_slice_header error
[h264 @ 00000000003e7440] no frame!
[h264 @ 00000000003e7440] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 00000000003e7440] decode_slice_header error
[h264 @ 00000000003e7440] no frame!
[h264 @ 00000000003e7440] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 00000000003e7440] decode_slice_header error
[h264 @ 00000000003e7440] no frame!
[h264 @ 00000000003e7440] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 00000000003e7440] decode_slice_header error
[h264 @ 00000000003e7440] no frame!
[h264 @ 00000000003e7440] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 00000000003e7440] decode_slice_header error
[h264 @ 00000000003e7440] no frame!
[mpegts @ 00000000003d2880] Could not find codec parameters for stream 2 (Unknown: none ([128][0][0][0] / 0x0080)): unknown codec
Consider increasing the value for the 'analyzeduration' and 'probesize' options
Input #0, mpegts, from 'udp://10.5.5.100:8554':
  Duration: N/A, start: 6.677333, bitrate: N/A
  Program 1
    Stream #0:0[0x1011]: Video: h264 (Main) ([27][0][0][0] / 0x001B), yuvj420p(pc, bt709, progressive), 848x480 [SAR 1:1 DAR 53:30], 29.97 fps, 29.97 tbr, 90k tbn, 59.94 tbc
    Stream #0:1[0x1100]: Audio: aac (LC) ([15][0][0][0] / 0x000F), 48000 Hz, stereo, fltp, 190 kb/s
[... 8 lines cut ...]

No more combinations to try, audio open failed
[h264 @ 0000000005aae540] concealing 917 DC, 917 AC, 917 MV errors in P frame
[swscaler @ 00000000062b1e40] deprecated pixel format used, make sure you did set range correctly
[h264 @ 0000000005aae540] error while decoding MB 42 28, bytestream -8
[h264 @ 0000000005aae540] concealing 113 DC, 113 AC, 113 MV errors in P frame
[h264 @ 0000000005aae540] concealing 761 DC, 761 AC, 761 MV errors in P frame
[h264 @ 0000000005aae540] concealing 251 DC, 251 AC, 251 MV errors in P frame
[h264 @ 0000000005aae540] cabac decode of qscale diff failed at 9 17
[h264 @ 0000000005aae540] error while decoding MB 9 17, bytestream 1724
[h264 @ 0000000005aae540] concealing 729 DC, 729 AC, 729 MV errors in P frame
[h264 @ 0000000005aae540] cabac decode of qscale diff failed at 49 16
[h264 @ 0000000005aae540] error while decoding MB 49 16, bytestream 1814
[h264 @ 0000000005aae540] concealing 742 DC, 742 AC, 742 MV errors in P frame
[h264 @ 0000000005aae540] cabac decode of qscale diff failed at 3 18
[h264 @ 0000000005aae540] error while decoding MB 3 18, bytestream 1506
[h264 @ 0000000005aae540] concealing 682 DC, 682 AC, 682 MV errors in P frame
[h264 @ 0000000005aae540] cabac decode of qscale diff failed at 10 10
[h264 @ 0000000005aae540] error while decoding MB 10 10, bytestream 2582
[h264 @ 0000000005aae540] concealing 1099 DC, 1099 AC, 1099 MV errors in P frame
[h264 @ 0000000005aae0c0] concealing 946 DC, 946 AC, 946 MV errors in P frame
  22.08 M-V: -0.908 fd=   0 aq=    0KB vq=    0KB sq=    0B f=0/0 
```

Is it something related to how i need to start ffplay ?

pderian, 2018-07-02:
Not sure, to be honest... It happens as well with my setup, I do not think it comes from ffplay.

I use a GoPro HERO 5 Black with firmware HD5.02.02.60.00.
I also noticed that the stream video freezes after some time (25 s I would say). In my use case it is not really a problem, as I stop/restart the stream regularly. I would feel better with another solution, however!

Regarding the ffplay output: the only thing I notice is that the audio (`Stream #0:1[0x1100]: Audio`) is not decoded. In my case I always disable it (`-na` option of ffplay) so I can't help with that. I'm not sure audio is supposed to be available anyway? The third stream (`Stream #0:2[0x200]: Unknown`) is probably the stream of metadata (gmpf), it is not surprising that ffplay does not know what to do with it.

pderian, 2018-07-04:
The solution was simple... we just forgot to run `GoPro.KeepAlive()` method to keep the camera alive while streaming.

Now, this function is blocking, so I would say the best approach is to run it in a separate thread.
Here is my version, which works fine in my case:
```
import threading
from goprocam import GoProCamera

def lifeguard(gpcam):
    """Creates a "lifeguard" thread tasked with keeping the camera alive e.g. for streaming.
    
    :param gpcam: a GoProCamera.GoPro instance;
    :return: a deamon thread running gpcam.KeepAlive(). 
    
    Written by P. DERIAN 2018-07-04.
    """
    t = threading.Thread(target=GoProCamera.GoPro.KeepAlive, args=(gpcam,), daemon=True)
    t.start()
    return t

# start the camera
gpcam = GoProCamera.GoPro()    
# create a lifeguard
lg = lifeguard(gpcam)

# now you can start the stream and do whatever you want.
# the lg thread will be terminated upon exit of the program.
```

zaidshakil, 2019-04-25:
Hello.. is there anyway i can record and stream at the same time?
When i send a record command to the camera, the streaming freezes!

KonradIT, 2019-04-25:
No.