KonradIT/gopro-py-api #48: Cannot install goprocam via PIP

xoul, 2018-07-07:
Cannot install goprocam via PIP because of missing README.md.

```
# pip install goprocam
Collecting goprocam
  Using cached https://files.pythonhosted.org/packages/bf/eb/8a7ac4b170ab95dfe13862b0ae73851116ee5d071e471d32899db2f08213/goprocam-2.0.5.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-KyU8Yu/goprocam/setup.py", line 2, in <module>
        with open("README.md", "r") as fh:
    IOError: [Errno 2] No such file or directory: 'README.md'
```

KonradIT, 2018-07-07:
https://github.com/KonradIT/gopro-py-api/releases/tag/2.0.6

xoul, 2018-07-07:
Tested and confirmed as working!

closed 2018-07-07 by KonradIT