KonradIT/gopro-py-api #50: Install issues

dcurzon1973, 2018-07-24:
I can connect my rPi to the GoPro wifi, and send commands via curl http://10.5.5.9/.....  but, the wifi times out on the gopro after about 5 minutes.  So i thought that the gopro-py-api might be the way to go.

cloned it to the Pi, but fails on running setup.py on line 1.  Something about setuptools.  So i installed setuptools, but still errors.  The Pi has Raspbian Stretch Lite installed and  Python3.
I'm pretty certain that it will be something that i'm doing wrong.

My goal is simply to be able to send a shutter activation command to the gopro in burst mode 10 frames/3 seconds).
Where am i going wrong?

Many thanks

dcurzon1973, 2018-07-25:
Hi KonradIT
I've made some progress!  I had to install pip3 and then install.
Python is a bit of a learning curve... so I was going to run the Motion app with a pi-camera, and on movement detect, send a shutter command.
Progress so far... connect to GoPro wifi, run the keepalive.py (didn't install ffmpeg but its only need to run in the background to keep the wifi on and it seems to do that)
and then, can open another terminal and either python in and send commands... running an active session and commands are quick, but calling a script seems to take about 7 seconds?  alternatively, keep alive in the background and then curl http commands.
thanks

Lalorrs, 2019-08-08:
Hi, im trying to install goprocam but i get this error:

ERROR: Command errored out with exit status 1:
     command: 'c:\users\eduardo\appdata\local\programs\python\python37-32\python.exe' -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\Public\\Documents\\Wondershare\\CreatorTemp\\pip-install-u2722tj_\\goprocam\\setup.py'"'"'; __file__='"'"'C:\\Users\\Public\\Documents\\Wondershare\\CreatorTemp\\pip-install-u2722tj_\\goprocam\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base pip-egg-info
         cwd: C:\Users\Public\Documents\Wondershare\CreatorTemp\pip-install-u2722tj_\goprocam\
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\Public\Documents\Wondershare\CreatorTemp\pip-install-u2722tj_\goprocam\setup.py", line 5, in <module>
        with open("version.json", "r") as ver:
    FileNotFoundError: [Errno 2] No such file or directory: 'version.json'
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

What can i do? thanks

Timvrakas, 2019-08-20:
Seconded on above error, will try from git

KonradIT, 2019-08-20:
Fixed in 3.0.9