KonradIT/gopro-py-api #52: 500 ISE on attempt to access GoPro URLs

afogal, 2018-08-07:
The unit is a GoPro HERO4, running the AFSP firmware and an override.sh that changes the IP address and forces it onto channel 1. 

The camera runs fine for a long time, until something causes it to break. Then it will not respond to commands, and needs to be restarted. Trying to visit 10.6.5.9/ works, 10.6.5.9/go/gpStatus and similar all result in 500 internal server error. 

Any ideas?

KonradIT, 2018-08-07:
Hi, that's strange, I don't have a Hero 4 to test. Make sure the URL is /gp/gpControl/status

afogal, 2018-08-07:
It is difficult to reproduce. I copied the URL from another issue thread so it would've been correct when I tried it. 

No ideas what might cause a 500 error code? Maybe the gpcontrol backend crashes?

KonradIT, 2018-08-07:
Try issuing the command from a ssh session, I think curl is installed. Also check dmesg for any details

KonradIT, 2018-08-07:
My bad it's Telnet, not ssh: https://3drpilots.com/threads/how-to-connect-to-a-gopro-hero-4-black-and-get-a-root-shell.3610/

afogal, 2018-08-10:
I couldn't find any relevant details, is there any way to restart it in place without physical access to the cameras?