KonradIT/gopro-py-api #53: API hung after Take Photo/Video

ermuir, 2018-08-09:
KonradIT,

First, let me commend you on the GoPro Python API.  It has some great functionality.

One thing I noticed today is that the API seems to get hung after issuing take_photo() or take_video(10):

**>>> gopro = GoProCamera.GoPro()
HERO5 Black
HD5.02.02.60.00
Camera successfully connected!
>>> gopro.take_photo()
{}

{}

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/site-packages/goprocam/GoProCamera.py", line 430, in take_photo
    ready=int(self.getStatus(constants.Status.Status, constants.Status.STATUS.IsBusy))
ValueError: invalid literal for int() with base 10: ''
>>>** 

I received the Traceback after issuing a Ctrl+C.  From the GoPro screen, it appears the camera has finished taking the photo/video and returns to standby.  So this may just be an issue with the IsBust status.

Do you have any suggestions?

Thanks in advance!

pderian, 2018-08-09:
> File "", line 1, in

Hi,

Jumping in as I faced the same problem.  I'm on vacations so I can't look
at my code, but here's what I recall.

Sometimes the result of `getStatus` cannot be interpreted as an `int`,
perhaps the value wasn't set internally when the call is made and the
camera returns ''. This is what happened according to your trace.

What I did is simply wrap a `try` / `except ValueError` around the call to
`reday = int(getStatus(...))`. If the ValueError is raised, then it means
the status code is not reliable, and you can return wathever fits your
needs.

Cheers,
Pierre

KonradIT, 2018-08-09:
Hi @ermuir @pderian ,

I ran the code just to test, it does return an URL in my case. (same camera and same firmware).

````
Python 3.7.0 (default, Jul 15 2018, 10:44:58) 
[GCC 8.1.1 20180531] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from goprocam import GoProCamera, constants
>>> gopro = GoProCamera.GoPro()
HERO5 Black
HD5.02.02.60.00
Camera successfully connected!
>>> gopro.take_photo()
{}

{}

'http://10.5.5.9:8080/videos/DCIM/102GOPRO/GOPR1824.JPG'

````

I can't reproduce it, I've tried it multiple times. What SD card are you using?

KonradIT, 2018-08-09:
Also run `gopro.getStatusRaw()` for debugging, paste the output here.

ermuir, 2018-08-09:
pderian,

I thought about using try/except as a work around.  I will give that a try if I am not able to find a solution.

KonradIT,

The SD card is a SanDisk Extreme 64GB microSD card.

Here is the output you requested:

>>> gopro.getStatusRaw()
'{"status":{\n"1":1,"2":2,"3":0,"4":0,"6":0,"8":1,"9":0,"10":0,"11":0,"13":0,"14":0,"15":0,"16":0,"17":0,"19":1,"20":1,"21":0,"22":0,"23":0,"24":0,"26":0,"27":0,"28":0,"29":"","30":"GP54691773","31":0,"32":0,"33":0,"34":9241,"35":14551,"36":22,"37":6,"38":110,"39":6,"40":"%12%08%09%07%03%22","41":0,"42":0,"43":0,"44":0,"45":0,"46":1,"47":1,"48":1,"49":0,"54":61503616,"55":1,"56":4,"57":90016,"58":0,"59":0,"60":500,"61":2,"62":0,"63":0,"64":0,"65":0,"66":0,"67":0,"68":1,"69":1,"70":63,"71":12,"72":19,"73":20,"74":0},\n"settings":{"1":0,"2":9,"3":10,"4":4,"5":0,"6":1,"7":1,"8":1,"9":0,"10":0,"11":0,"12":0,"13":1,"14":0,"15":4,"16":0,"17":10,"18":4,"19":0,"20":0,"21":0,"22":0,"23":0,"24":4,"25":0,"26":4,"27":0,"28":0,"29":5,"30":0,"31":0,"32":3601,"33":0,"34":0,"35":0,"36":0,"37":4,"38":0,"39":4,"40":0,"41":13,"42":8,"43":0,"44":8,"45":8,"46":0,"47":0,"48":0,"50":0,"51":1,"52":0,"54":1,"57":0,"58":1,"59":6,"60":8,"61":1,"62":2500000,"63":0,"64":4,"65":0,"66":0,"67":0,"68":0,"69":1,"70":0,"71":0,"72":1,"73":0,"74":0,"75":3,"76":3,"77":0,"78":1,"79":0,"80":2,"81":3,"82":0,"83":1,"84":0,"85":0,"86":0,"87":0,"88":100,"89":12,"91":2,"92":12,"93":0,"94":0,"95":1,"96":0,"97":0,"98":0,"99":0}}\n'

Thanks!

KonradIT, 2018-08-09:
@emuir was the camera recording when you ran getStatusRaw()?

ermuir, 2018-08-09:
No, it was just in standby. Should I initiate video recording through the
GoPro itself and then issue the getStatusRaw() command?

KonradIT, 2018-08-09:
Yes

ermuir, 2018-08-09:
Here is the getStatusRaw() output from a separate terminal after issuing the take_photo() command, which was still hung:

>>> gopro.getStatusRaw()
'{"status":{\n"1":1,"2":2,"3":0,"4":0,"6":0,"8":1,"9":0,"10":0,"11":0,"13":0,"14":0,"15":0,"16":0,"17":0,"19":1,"20":1,"21":0,"22":0,"23":0,"24":0,"26":0,"27":0,"28":0,"29":"","30":"GP54691773","31":0,"32":0,"33":0,"34":24024,"35":14550,"36":23,"37":6,"38":111,"39":6,"40":"%12%08%09%07%30%14","41":0,"42":0,"43":1,"44":1,"45":0,"46":1,"47":1,"48":1,"49":0,"54":61500672,"55":1,"56":4,"57":385748,"58":0,"59":0,"60":500,"61":2,"62":0,"63":0,"64":0,"65":0,"66":0,"67":0,"68":1,"69":1,"70":53,"71":12,"72":17,"73":20,"74":0},\n"settings":{"1":0,"2":9,"3":10,"4":4,"5":0,"6":1,"7":1,"8":1,"9":0,"10":0,"11":0,"12":0,"13":1,"14":0,"15":4,"16":0,"17":10,"18":4,"19":0,"20":0,"21":0,"22":0,"23":0,"24":4,"25":0,"26":4,"27":0,"28":0,"29":5,"30":0,"31":0,"32":3601,"33":0,"34":0,"35":0,"36":0,"37":4,"38":0,"39":4,"40":0,"41":13,"42":8,"43":0,"44":8,"45":8,"46":0,"47":0,"48":0,"50":0,"51":1,"52":0,"54":1,"57":0,"58":1,"59":6,"60":8,"61":1,"62":2500000,"63":0,"64":4,"65":0,"66":0,"67":0,"68":0,"69":1,"70":0,"71":0,"72":1,"73":0,"74":0,"75":3,"76":3,"77":0,"78":1,"79":0,"80":2,"81":3,"82":0,"83":1,"84":0,"85":0,"86":0,"87":0,"88":100,"89":12,"91":2,"92":17,"93":0,"94":0,"95":1,"96":0,"97":0,"98":0,"99":0}}\n'

This is the status after Ctrl+C:

'{"status":{\n"1":1,"2":2,"3":0,"4":0,"6":0,"8":1,"9":0,"10":0,"11":0,"13":0,"14":0,"15":0,"16":0,"17":0,"19":1,"20":1,"21":0,"22":0,"23":0,"24":0,"26":0,"27":0,"28":0,"29":"","30":"GP54691773","31":0,"32":0,"33":0,"34":9233,"35":14539,"36":23,"37":7,"38":111,"39":7,"40":"%12%08%09%07%31%0A","41":0,"42":0,"43":0,"44":0,"45":0,"46":1,"47":1,"48":1,"49":0,"54":61453056,"55":1,"56":4,"57":435764,"58":0,"59":0,"60":500,"61":2,"62":0,"63":0,"64":0,"65":0,"66":0,"67":0,"68":1,"69":1,"70":53,"71":12,"72":17,"73":20,"74":0},\n"settings":{"1":0,"2":9,"3":10,"4":4,"5":0,"6":1,"7":1,"8":1,"9":0,"10":0,"11":0,"12":0,"13":1,"14":0,"15":4,"16":0,"17":10,"18":4,"19":0,"20":0,"21":0,"22":0,"23":0,"24":4,"25":0,"26":4,"27":0,"28":10,"29":5,"30":0,"31":0,"32":3601,"33":0,"34":0,"35":0,"36":0,"37":0,"38":0,"39":4,"40":0,"41":13,"42":8,"43":0,"44":8,"45":8,"46":0,"47":0,"48":0,"50":0,"51":1,"52":0,"54":1,"57":0,"58":1,"59":6,"60":8,"61":1,"62":2500000,"63":0,"64":4,"65":0,"66":0,"67":0,"68":0,"69":1,"70":1,"71":0,"72":1,"73":0,"74":0,"75":3,"76":1,"77":0,"78":1,"79":0,"80":2,"81":3,"82":0,"83":1,"84":0,"85":0,"86":0,"87":0,"88":100,"89":12,"91":2,"92":12,"93":0,"94":0,"95":1,"96":0,"97":0,"98":0,"99":0}}\n'

And finally, this is the status while the camera is video recording (initiated from the GoPro):

'{"status":{\n"1":1,"2":2,"3":0,"4":0,"6":0,"8":1,"9":0,"10":1,"11":0,"13":7,"14":0,"15":0,"16":0,"17":0,"19":1,"20":1,"21":0,"22":0,"23":0,"24":0,"26":0,"27":0,"28":0,"29":"","30":"GP54691773","31":0,"32":0,"33":0,"34":9240,"35":14550,"36":23,"37":6,"38":111,"39":6,"40":"%12%08%09%07%30%32","41":0,"42":0,"43":0,"44":0,"45":0,"46":1,"47":1,"48":1,"49":0,"54":61470579,"55":1,"56":4,"57":416390,"58":0,"59":0,"60":500,"61":0,"62":0,"63":0,"64":0,"65":0,"66":0,"67":0,"68":1,"69":1,"70":53,"71":12,"72":17,"73":20,"74":0},\n"settings":{"1":0,"2":9,"3":10,"4":4,"5":0,"6":1,"7":1,"8":1,"9":0,"10":0,"11":0,"12":0,"13":1,"14":0,"15":4,"16":0,"17":10,"18":4,"19":0,"20":0,"21":0,"22":0,"23":0,"24":4,"25":0,"26":4,"27":0,"28":10,"29":5,"30":0,"31":0,"32":3601,"33":0,"34":0,"35":0,"36":0,"37":0,"38":0,"39":4,"40":0,"41":13,"42":8,"43":0,"44":8,"45":8,"46":0,"47":0,"48":0,"50":0,"51":1,"52":0,"54":1,"57":0,"58":1,"59":6,"60":8,"61":1,"62":2500000,"63":0,"64":4,"65":0,"66":0,"67":0,"68":0,"69":1,"70":1,"71":0,"72":1,"73":0,"74":0,"75":3,"76":1,"77":0,"78":1,"79":0,"80":2,"81":3,"82":0,"83":1,"84":0,"85":0,"86":0,"87":0,"88":100,"89":12,"91":2,"92":12,"93":0,"94":0,"95":1,"96":0,"97":0,"98":0,"99":0}}\n'

KonradIT, 2018-08-09:
Restart your camera. Try connecting it to the GoPro app and see if you can browse media.

dnc1dr, 2018-09-04:
Jumping in -- I've been experiencing the same with my Hero4.

olegdatskiv, 2019-06-29:
Hi KonradIT,

I have similar problem with shoot_video() method. How can this be resolved? 

Thanks,
Oleh.