KonradIT/gopro-py-api #58: import goprocam error

gbaweja, 2018-09-13:
it says,

ImportError: No Module named request

KonradIT, 2018-09-16:
Library works with Python 3.6 and above.

closed 2018-09-16 by KonradIT

tacleur, 2019-01-17:
Hello everyone,

I worked for 4 months for make video and photo capture with your API but the problem is that python don't recognize goprocam. My error message : 

`Python 3.7.2 (default, Jan 17 2019, 14:20:11) 
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 
>>> 
>>> from goprocam import GoProCamera, constants
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'goprocam'
>>> 
`
Can someone help me ?

Thanks a lot !

KonradIT, 2019-01-17:
You don't have goprocam installed.

tacleur, 2019-01-18:
Hi,

Thanks for your answer.
I had installed several times your API but i have this message.
I had follow your install procedure and the terminal said sucessful.
(I'm working on Linux Raspberry pi).

Thanks for your help.

KonradIT, 2019-01-18:
Even pip install goprocam?

tacleur, 2019-01-18:
> 
> 
> Even pip install goprocam?

yes.
I'll do it again : 

`pi@pi:~ $ sudo pip install goprocam
Requirement already satisfied: goprocam in /usr/local/lib/python2.7/dist-packages/goprocam-3.0.1-py2.7.egg
pi@pi:~ $ pip install goprocam
Collecting goprocam
Installing collected packages: goprocam
Successfully installed goprocam-3.0.3
pi@pi:~ $ from goprocam import GoProCamera, constants
from: can't read /var/mail/goprocam
pi@pi:~ $ python3
Python 3.7.2 (default, Jan 17 2019, 14:20:11) 
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from goprocam import GoProCamera, constants
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'goprocam'
>>> 

`thanks for your help

gbaweja, 2019-01-18:
> File "", line 1, in

you need to do..

sudo pip3 install goprocam

tacleur, 2019-01-22:
hi,
thanks for your answer.

Unfortunately, it don't works. impossible ton install with pip3 : 

`pi@pi:~ $ sudo pip3 install goprocam
pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting goprocam
  Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.",)': /simple/goprocam/
Could not fetch URL https://pypi.org/simple/goprocam/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/goprocam/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.",)) - skipping
 Could not fetch URL https://www.piwheels.org/simple/goprocam/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='www.piwheels.org', port=443): Max retries exceeded with url: /simple/goprocam/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.",)) - skipping
  Could not find a version that satisfies the requirement goprocam (from versions: )
No matching distribution found for goprocam
pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.
Could not fetch URL https://pypi.org/simple/pip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/pip/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.",)) - skipping
Could not fetch URL https://www.piwheels.org/simple/pip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='www.piwheels.org', port=443): Max retries exceeded with url: /simple/pip/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.",)) - skipping`

How pass this problem?

Thansk for your help;

KonradIT, 2019-01-22:
Seems like the SSL is badly configured.

tacleur, 2019-01-26:
How can i correct SSL ?

tacleur, 2019-01-27:
I found the problem.
I erased all my raspberry and now it's works with pip3.