KonradIT/gopro-py-api #61: error running simple code

iacob69, 2018-10-02:
Hi Konrad,

I've got a Hero3 connected to the pc (Win10) with Python 3.7 and when I run:
from goprocam import GoProCamera
from goprocam import constants

gpCam = GoProCamera.GoPro()

I get this error:
Traceback (most recent call last):
  File "C:\Users\iacob\Desktop\PYTHONproj\gopro\gopro1.py", line 1, in <module>
    from goprocam import GoProCamera
  File "C:\Users\iacob\AppData\Local\Programs\Python\Python37\lib\site-packages\goprocam\__init__.py", line 1, in <module>
    from goprocam import GoProCamera
  File "C:\Users\iacob\AppData\Local\Programs\Python\Python37\lib\site-packages\goprocam\GoProCamera.py", line 51
    except ImportError, e:
                      ^
SyntaxError: invalid syntax
>>> 
What should I do?

Thanks!

Iacob

KonradIT, 2018-10-02:
Fixed. Run `git pull` if you're running from source or run `pip install goprocam` to update from Pypi.

iacob69, 2018-10-02:
Thanks,
Now I get this error:
Traceback (most recent call last):
  File "C:\Users\iacob\Desktop\PYTHONproj\gopro\gopro1.py", line 1, in <module>
    from goprocam import GoProCamera
  File "C:\Users\iacob\AppData\Local\Programs\Python\Python37\lib\site-packages\goprocam\__init__.py", line 1, in <module>
    from goprocam import GoProCamera
  File "C:\Users\iacob\AppData\Local\Programs\Python\Python37\lib\site-packages\goprocam\GoProCamera.py", line 488
    self.gpControlCommand("digital_zoom?range_pcnt=" + zoomLevel)
       ^
IndentationError: expected an indented block

iacob69, 2018-10-02:
Fixed it:
Just edited GoProCamera.py until I got this:
HTTP/1.1 ÿûAllow: GET 

None

KonradIT, 2018-10-02:
Yes that's what you should get.

iacob69, 2018-10-09:
Thanks a lot.
Now, when I run this:
from goprocam import GoProCamera
from goprocam import constants

gpCam = GoProCamera.GoPro()
gpCam.pair(usepin=False)
gpCam.overview()
gpCam.getStatusRaw()
gpCam.take_photo(5)
gpCam.downloadLastMedia()

I get this error:

HERO
H18.01.01.00.00
Camera successfully connected!
Paired
camera overview
current mode: Photo
current submode: Burst
current video resolution: 1080p
current video framerate: 25
pictures taken: 25
videos taken:  0
videos left: 01:00:55
pictures left: 6037
battery left: Full
space left in sd card: 14.39GB
camera SSID: GP24707879
Is Recording: Not recording - standby
Clients connected: 0
camera model: HERO
firmware version: H18.01.01.00.00
serial number: C3321324707879
{}

wait 5 seconds.
{}

Traceback (most recent call last):
  File "C:\Users\iacob\Desktop\PYTHONproj\gopro\gopro4.py", line 8, in <module>
    gpCam.take_photo(5)
  File "C:\Users\iacob\AppData\Local\Programs\Python\Python37\lib\site-packages\goprocam\GoProCamera.py", line 446, in take_photo
    return self.getMedia()
AttributeError: 'GoPro' object has no attribute 'getMedia'
>>> 

Could you please kindly help?

KonradIT, 2018-10-09:
Sorry. fixed.

iacob69, 2018-10-09:
Thanks but Error remains after reinstalling goprocam
HERO
H18.01.01.00.00
Camera successfully connected!
Paired
camera overview
current mode: Photo
current submode: Burst
current video resolution: 1080p
current video framerate: 25
pictures taken: 47
videos taken:  0
videos left: 01:00:43
pictures left: 6016
battery left: Halfway
space left in sd card: 14.34GB
camera SSID: GP24707879
Is Recording: Not recording - standby
Clients connected: 0
camera model: HERO
firmware version: H18.01.01.00.00
serial number: C3321324707879
{}

{}

Traceback (most recent call last):
  File "C:/Users/iacob/Desktop/PYTHONproj/gopro/gopro5.py", line 8, in <module>
    gpCam.take_photo(1)
  File "C:\Users\iacob\AppData\Local\Programs\Python\Python37\lib\site-packages\goprocam\GoProCamera.py", line 446, in take_photo
    return self.getMedia()
AttributeError: 'GoPro' object has no attribute 'getMedia'
>>>

iacob69, 2018-10-09:
Works now, needed to install, By the Way camera is GoPro2018
Many thanks! Great Job!