KonradIT/gopro-py-api #65: pip version - getMedia function partially false indented

maxstrobel, 2018-10-17:
**Problem:** Parts of the getMedia function are false indented in the version installed by pip

**Details:** Lines 492-494 false indented

```
        ##
                def getMedia(self):
                        folder = ""
                        file_lo = ""
                try:
```

* GoPro Camera(s): -
* Firmware Version: -
* Steps to reproduce:
```
max@MaxKubuntuNUC:~$ conda create -n GoPro-Test python=3.6 pip            
Solving environment: done

## Package Plan ##

  environment location: /home/max/miniconda3/envs/GoPro-Test

  added / updated specs: 
    - pip
    - python=3.6

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    readline-7.0               |       haf1bffa_1         381 KB  conda-forge
    python-3.6.6               |       h5001a0f_3        20.5 MB  conda-forge
    libgcc-ng-7.2.0            |       hdf63c60_3         6.1 MB  conda-forge
    sqlite-3.25.2              |       hb1c47c0_0         1.6 MB  conda-forge
    xz-5.2.4                   |       h470a237_1         328 KB  conda-forge
    libffi-3.2.1               |       hfc679d8_5          51 KB  conda-forge
    tk-8.6.8                   |       ha92aebf_0         3.1 MB  conda-forge
    setuptools-40.4.3          |           py36_0         554 KB  conda-forge
    ncurses-6.1                |       hfc679d8_1         1.2 MB  conda-forge
    wheel-0.32.1               |           py36_0          34 KB  conda-forge
    zlib-1.2.11                |       h470a237_3          93 KB  conda-forge
    pip-18.1                   |        py36_1000         1.8 MB  conda-forge
    libstdcxx-ng-7.2.0         |       hdf63c60_3         2.5 MB  conda-forge
    ------------------------------------------------------------
                                           Total:        38.3 MB

The following NEW packages will be INSTALLED:

    ca-certificates: 2018.10.15-ha4d7672_0 conda-forge
    certifi:         2018.10.15-py36_1000  conda-forge
    libffi:          3.2.1-hfc679d8_5      conda-forge
    libgcc-ng:       7.2.0-hdf63c60_3      conda-forge
    libstdcxx-ng:    7.2.0-hdf63c60_3      conda-forge
    ncurses:         6.1-hfc679d8_1        conda-forge
    openssl:         1.0.2p-h470a237_1     conda-forge
    pip:             18.1-py36_1000        conda-forge
    python:          3.6.6-h5001a0f_3      conda-forge
    readline:        7.0-haf1bffa_1        conda-forge
    setuptools:      40.4.3-py36_0         conda-forge
    sqlite:          3.25.2-hb1c47c0_0     conda-forge
    tk:              8.6.8-ha92aebf_0      conda-forge
    wheel:           0.32.1-py36_0         conda-forge
    xz:              5.2.4-h470a237_1      conda-forge
    zlib:            1.2.11-h470a237_3     conda-forge

Proceed ([y]/n)? y

Downloading and Extracting Packages
readline-7.0         | 381 KB    | ####################################################################################### | 100% 
python-3.6.6         | 20.5 MB   | ####################################################################################### | 100% 
libgcc-ng-7.2.0      | 6.1 MB    | ####################################################################################### | 100% 
sqlite-3.25.2        | 1.6 MB    | ####################################################################################### | 100% 
xz-5.2.4             | 328 KB    | ####################################################################################### | 100% 
libffi-3.2.1         | 51 KB     | ####################################################################################### | 100% 
tk-8.6.8             | 3.1 MB    | ####################################################################################### | 100% 
setuptools-40.4.3    | 554 KB    | ####################################################################################### | 100% 
ncurses-6.1          | 1.2 MB    | ####################################################################################### | 100% 
wheel-0.32.1         | 34 KB     | ####################################################################################### | 100% 
zlib-1.2.11          | 93 KB     | ####################################################################################### | 100% 
pip-18.1             | 1.8 MB    | ####################################################################################### | 100% 
libstdcxx-ng-7.2.0   | 2.5 MB    | ####################################################################################### | 100% 
Preparing transaction: done
Verifying transaction: done
Executing transaction: done
#
# To activate this environment, use:
# > source activate GoPro-Test
#
# To deactivate an active environment, use:
# > source deactivate
#
max@MaxKubuntuNUC:~$ source activate GoPro-Test
(GoPro-Test) max@MaxKubuntuNUC:~$ pip install goprocam
Collecting goprocam
Installing collected packages: goprocam
Successfully installed goprocam-2.0.9.post1
(GoPro-Test) max@MaxKubuntuNUC:~$ sed -n '486,511p;512q' ~/miniconda3/envs/GoPro-Test/lib/python3.6/site-packages/goprocam/GoProCamera.py 
        def setZoom(self, zoomLevel):
                if zoomLevel >= 0 and zoomLevel <= 100:
                        self.gpControlCommand("digital_zoom?range_pcnt=" + zoomLevel)
        ##
        ## Query media list
        ##
                def getMedia(self):
                        folder = ""
                        file_lo = ""
                try:
                        raw_data = urllib.request.urlopen('http://' + self.ip_addr + ':8080/gp/gpMediaList').read().decode('utf-8')
                        json_parse = json.loads(raw_data)
                        for i in json_parse['media']:
                                folder=i['d']
                        for i in json_parse['media']:
                                for i2 in i['fs']:
                                        file_lo = i2['n']
                        return "http://" + self.ip_addr + ":8080/videos/DCIM/" + folder + "/" + file_lo
                except (HTTPError, URLError) as error:
                        return ""
                        print("Error code:" + str(error.code) + "\nMake sure the connection to the WiFi camera is still active.")
                except timeout:
                        return ""
                        print("HTTP Timeout\nMake sure the connection to the WiFi camera is still active.")
        def getMediaFront(self):
                folder = ""
(GoPro-Test) max@MaxKubuntuNUC:~$
```

* Happens every time? [Y/N]: Y

KonradIT, 2018-10-17:
Roger that. Fixing it now.

KonradIT, 2018-10-17:
Should work with 2.0.9.post1 version. 3.0.0 is now in pip.

closed 2018-10-17 by KonradIT