KonradIT/gopro-py-api #66: Hero7 support

var316, 2018-10-17:
Hi Konrad, all, 

Has anyone made it work with Hero 7 Black or the support is still in progress? 

Cheers

KonradIT, 2018-10-17:
It works with the 7 black. I've tested it myself.

var316, 2018-10-18:
I've been trying to connect to the camera through the 'gpCam = GoProCamera.GoPro(constants.gpcontrol)' command with no success and stuck in 'Waking up...' state. 

And next I tried to connect by setting the mac_address through 'GoProCamera.GoPro("DC:53:60:20:EF:7C", constants.gpcontrol)' but this returns another error mentioned below:

[image]

Can you please tell me where am I going wrong?

KonradIT, 2018-10-18:
Hi

Try with: GoProCamera.GoPro(constants.gpcontrol,  mac_address="DC:53:60:20:EF:7C")

var316, 2018-10-18:
I tried with that and also with: 
GoProCamera.GoPro(constants.gpcontrol, ip_address="10.5.5.101", mac_address="DC:53:60:20:EF:7C")
but it still is stuck in the Waking Up state and eventually the camera wifi also disappears. Did you also face such problems with the GoPro wifi?

Many thanks for your support btw!

KonradIT, 2018-10-18:
That's weird. Try with "dc536020ef7c" or install "getmac" module.

KonradIT, 2018-10-18:
Proof it works with my HERO7 Black: https://streamable.com/v1879

var316, 2018-10-18:
Hey, it works! I had changed the ip_address to the laptop ip_address (annoyingly) in the GoProCamera.py so it just works with the GoProCamera.GoPro(constants.gpcontrol) command itself.

KonradIT, 2018-10-18:
Sweet.

closed 2018-10-18 by KonradIT