KonradIT/gopro-py-api #67: Recording Timelapse Video

var316, 2018-11-02:
Hi Konrad,

I'm trying to record a timelapse video but something seems not right. I first set the Timelapse Video mode through this:
gopro = GoProCamera.GoPro(constants.gpcontrol)
gopro.mode(constants.Mode.VideoMode, constants.Mode.SubMode.Video.TimeLapseVideo),  constants.Video.VideoTimeLapseInterval.IHalf1)
gopro.gpControlSet(constants.Video.VIDEO_TIMELAPSE_INTERVAL, constants.Video.VideoTimeLapseInterval.IHalf1)

I tried using shoot_video after this but I think it sets to video mode. After that I tried 
 gopro.shutter(constants.start). 

Can you please help me with this? Also, how to set the timelapse video duration?

Thanks.

KonradIT, 2018-11-02:
Hi,

It should be:

````
gopro = GoProCamera.GoPro(constants.gpcontrol)
gopro.mode(constants.Mode.VideoMode, constants.Mode.SubMode.Video.TimeLapseVideo), 
gopro.gpControlSet(constants.Video.VIDEO_TIMELAPSE_INTERVAL, constants.Video.VideoTimeLapseInterval.IHalf1)
gopro.shutter(constants.start)
````

var316, 2018-11-05:
I had these commands as well but was not working before. Works now though, thanks!

closed 2018-11-05 by var316

var316, 2018-11-05:
Hi Konrad,

Can you please let me know what do the VideoTimeLapseInterval values indicate?
IHalf1 = 1 frame recorded every 0.5 sec
I1 = 1 frame/1 sec
I2 = 1 frame/2 sec 
I5 = 1 frame/5 sec
Is that correct?

KonradIT, 2018-11-05:
Yes.