KonradIT/gopro-py-api #68: Couldn't read a new frame

fc73, 2018-11-07:
Hi,
I met some problems when I was trying to get the frames from the live stream. Here is the error:
warning: Error opening file (/build/opencv/modules/videoio/src/cap_ffmpeg_impl.hpp:856)
warning: udp://10.5.5.9:8554 (/build/opencv/modules/videoio/src/cap_ffmpeg_impl.hpp:857)

It seems like there is no frame generated by the life stream since the images generated are always 0kb.
The running environment is python3.6 windows 7.

Any help is appreciated!

BR
Cheng

KonradIT, 2018-11-08:
Hey there, can you post the full code? What version of OpenCV do you have installed?

fc73, 2018-11-08:
I think I should implement it in Linux. It seems like hero5 can’t do the live streaming in windows.

KonradIT, 2018-11-08:
For windows I think the IP address needs to be specified in the file.