KonradIT/gopro-py-api #70: Issue with streamsettings

var316, 2018-11-22:
Hi,

I'm trying to change the live stream settings to a better quality but it does not seem to work.
I tried it in ffmpegless-preview.py by using:

,,,
gpCam = GoProCamera.GoPro()
gpCam.streamSettings(constants.Stream.BitRate.B1Mbps, constants.Stream.WindowSize.R720)
gpCam.livestream("start")
'''

This doesn't change anything as the 'frame' size at all times remains 480x640x3. 

Can someone please help with it?

Cheers

var316, 2018-11-22:
I am using the Hero7 black

KonradIT, 2018-11-22:
I get 848x480 [SAR 1:1 DAR 53:30], 29.97 fps on my HERO7 Black with the same code.

var316, 2018-11-22:
Here is the code I'm using. Is it the same code at your end?

```
import cv2
import numpy as np
from time import time
import socket
from goprocam import GoProCamera
from goprocam import constants

gpCam = GoProCamera.GoPro()
gpCam.streamSettings(constants.Stream.BitRate.B1Mbps, constants.Stream.WindowSize.R720)

sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
t = time()
gpCam.livestream("start")
cap = cv2.VideoCapture("udp://10.5.5.9:8554")
while True:
    nmat, frame = cap.read()
    cv2.imshow("GoPro OpenCV", frame)
    if cv2.waitKey(1) & 0xFF == ord('q'):
        break
    if time() - t >= 2.5:
        sock.sendto("_GPHD_:0:0:2:0.000000\n".encode(), ("10.5.5.9", 8554))
        t = time()
# When everything is done, release the capture
cap.release()
cv2.destroyAllWindows()
```

KonradIT, 2018-11-22:
I'm using FFplay, will try with OpenCV later

var316, 2018-11-22:
That would be great and thank you. Can you please share the ffplay command that you used?

KonradIT, 2018-11-22:
ffplay udp://10.5.5.9:8554

var316, 2018-11-23:
Thanks. Did you manage to make it work with OpenCV?