KonradIT/gopro-py-api #75: Retrieve raw photo via downloadLastMedia?

photonclock, 2018-12-30:
When Hero5 Black takes a raw photo, it records both file.GPR and file.JPG  

Could you add a method to retrieve both, or perhaps a parameter/option to retrieve either GPR or JPG?

PS - I really appreciate this API you've built. Thank you again!

DeivisPau, 2019-01-23:
I think that's the limitation of GoPro. It only allows raw to be exported via cable or card reader. 
Konrad, could you confirm that?

KonradIT, 2019-01-23:
The API can actually retrieve GPR files but I need to implement a separate command. Something like downloadGPR(last=False, folder, filename)

DeivisPau, 2019-01-23:
That's fantastic news. What would the URL call look like then to retrieve all raw files? This is the only missing piece to my time-lapse project. I wouldn't mind donating for you if you would implement it to your API!

KonradIT, 2019-01-23:
Since I've gotten requests for this feature I'll add it this afternoon. The function(s) should look like:

    downloadRawPhoto(folder, file, custom_filename="") #100GOPRO , GOPRXXXX.GPR optional: custom file name

    downloadLastRawPhoto(custom_filename="") #optional: custom file name

KonradIT, 2019-01-23:
Added new functions in 6a25cc8, untested till I get a chance. @DeivisPau do you mind testing them? 

downloadLastRawPhoto should work if the last media item taken is a photo and has RAW enabled.

downloadRawPhoto accepts JPG or GPR as parameters.

DeivisPau, 2019-01-23:
Thanks for this Konrad.

KonradIT, 2019-01-23:
Hi, try with 

    gp.downloadRawPhoto("100GOPRO", "G0025532.GPR")

DeivisPau, 2019-01-23:
That works like a charm.
Is there a way to download the whole folder (including raw files) or will I have to set up a loop to get them one by one?

KonradIT, 2019-01-23:
you can use downloadAll() which will download all media but won't download GPR files, to get GPR files you'd have to iterate like:

    for photoitem in gopro.listMedia(format=True, media_array=True):
        if photoitem[1].endswith("JPG"):
            try:
                gopro.downloadRawPhoto(photoitem[0], photoitem[1])
             catch:
                print("Not a raw photo")

Code is untested.

DeivisPau, 2019-01-23:
I've tried downloadAll() but it only downloaded a selection of files. Is there a reason for that?
The return I got:
```
>>> gp.downloadAll()
filename: GOPR5528.JPG
filename: G0015529.JPG
filename: G0025533.JPG
['GOPR5528.JPG', 'G0015529.JPG', 'G0025533.JPG']
>>>
```
But when i connect camera i can see 7 JPG files. listMedia() also returns only those three files.
[image]

KonradIT, 2019-01-23:
Wow that is weird, I'll check on that later when I have access to my camera.

In the meantime try:

    for photoitem in gopro.listMedia(format=True, media_array=True):
        if photoitem[1].endswith("JPG"):
            gopro.downloadMedia(photoitem[0], photoitem[1])
            try:
                gopro.downloadRawPhoto(photoitem[0], photoitem[1])
             catch:
                print("Not a raw photo")

DeivisPau, 2019-01-23:
It gives an error:

```
from goprocam import GoProCamera, constants
gp = GoProCamera.GoPro()

for photoitem in gp.listMedia(format=True, media_array=True):
    if photoitem[1].endswith("JPG"):
        try:
            gp.downloadRawPhoto(photoitem[0], photoitem[1])
        catch:
            print("Not a raw photo")
```
Error message:
```
    catch:
        ^
SyntaxError: invalid syntax
```

KonradIT, 2019-01-23:
Testing on my HERO7 Black

```
>>> media = gopro.listMedia(format=True, media_array=True)
>>> for i in media:
...     print(i)
... 
['100GOPRO', 'GOPR3149.JPG', '7426598', '1547999876']
['100GOPRO', 'GH013150.MP4', '29288750', '1547999940']
['100GOPRO', 'GOPR3151.JPG', '5416887', '1548000240']
['100GOPRO', 'GOPR3152.JPG', '4562748', '1548000248']
['100GOPRO', 'GOPR3153.JPG', '4490214', '1548000260']
['100GOPRO', 'GOPR3154.JPG', '5162273', '1548000264']
>>> gopro.downloadAll()
filename: GOPR3149.JPG
filename: GH013150.MP4
filename: GOPR3151.JPG
filename: GOPR3152.JPG
filename: GOPR3153.JPG
filename: GOPR3154.JPG
['GOPR3149.JPG', 'GH013150.MP4', 'GOPR3151.JPG', 'GOPR3152.JPG', 'GOPR3153.JPG', 'GOPR3154.JPG']

```

gopro.downloadAll() works perfectly for me. If not use the code I sent earlier which also downloads all the Raw Photos.

DeivisPau, 2019-01-23:
> Testing on my HERO7 Black
>
> ```
> ...     print(i)
> ...
> ['100GOPRO', 'GOPR3149.JPG', '7426598', '1547999876']
> ['100GOPRO', 'GH013150.MP4', '29288750', '1547999940']
> ['100GOPRO', 'GOPR3151.JPG', [...]

I think it only downloads the first image of the timelapse. Try shooting few timelapses and compare the images that are on SD card and via listMedia().

KonradIT, 2019-01-23:
Yes for timelapses you do have to use downloadMultishot().