KonradIT/gopro-py-api #78: AttributeError: 'GoPro' object has no attribute 'mac_address'

safeisnotanoption, 2019-01-14:
Hi Konrad,

I've got a Fusion connected to the pc (Win10) with Python 3.6.6 and when I run:

from goprocam import GoProCamera
from goprocam import constants

gpCam = GoProCamera.GoPro()

I get this error:

Traceback (most recent call last):
  File "D:\Python36_64\lib\site-packages\goprocam\GoProCamera.py", line 151, in whichCam
    response_raw = urllib.request.urlopen('http://' + self.ip_addr + '/gp/gpControl', timeout=5).read().decode('utf8')
  File "D:\Python36_64\lib\http\client.py", line 459, in read
    s = self.fp.read()
  File "D:\Python36_64\lib\socket.py", line 586, in readinto
    return self._sock.recv_into(b)
socket.timeout: timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "d:/Working/HeliCam_v20190113/Core/Fusion/Fusion.py", line 5, in <module>
    gpCam = GoProCamera.GoPro()
  File "D:\Python36_64\lib\site-packages\goprocam\GoProCamera.py", line 55, in __init__
    self._camera = self.whichCam()
  File "D:\Python36_64\lib\site-packages\goprocam\GoProCamera.py", line 185, in whichCam
    self.power_on(self.mac_address)
AttributeError: 'GoPro' object has no attribute 'mac_address'

This error appeared suddenly, before that the camera worked normally for several days.

safeisnotanoption, 2019-01-14:
I apologize for this thread, it seems that my python was cluttered with libraries. I reinstalled everything in a fresh virtual environment and it all worked.

KonradIT, 2019-01-14:
Hi, I'll look into this. Can you get the latest from source?

safeisnotanoption, 2019-01-14:
I apologize for this thread, it seems that my python was cluttered with libraries. I reinstalled everything in a fresh virtual environment and it worked.

closed 2019-01-14 by safeisnotanoption