KonradIT/gopro-py-api #81: OctoPi compatibility ?

yet-another-average-joe, 2019-02-03:
Hello,

I just spent 4 hours attempting to install and test gopro-py-api on a raspberry 3B+ running OctoPi/OctoPrint (a Jessie light distro running a 3D printing server). The goal is to shoot timelapses using a script called from a plugin (OctoLapse). It's easy to control a GoPro using the GPIO with some hack of   the shoot button of the GoPro or it's remote, or using a ESP8266 WiFi chip, but it would be nicer to do it through the integrated WiFi !

Something's wrong with my installation.

I installed Python3 (OctoPrint relies on Python 2.7), and pip3. gopro-py-api seems to install smoothly, but the first python3 command ("from goprocam import GoProCamera") fails with an error message "can't read /var/mail/goprocam". This folder just does not exist.

Something's probably missing somewhere with OctoPi distro , but what ???? And why /var/mail/goprocam directory ?

KonradIT, 2019-02-03:
Don't know how to debug this without a live board but from what I can gather try running a Python script with your commands and the proper shebang at the top.

yet-another-average-joe, 2019-02-03:
Thanks for your quick answer !

Now, I cd'ed to /home/pi/gopro-py-api/goprocam before I launched python3. Now I get :

File "<stdin>", line 1, in <module>
ImportError: No module named 'goprocam'

Such errors appears on one of your Vimeo tutorials...

I know what stdin is, at least in C/C++ language (the only one I really know and love, for 30+ years...). But it doen't help much. I wanted to learn Python with a real life project (not a blink pgm), but I feel like it will be the (very) hard way ! I understand that Git is for Bug Reports, Feature Requests and Pull Requests : is there an official thread for your library on some forum ?( I'm a Windows APi fanboy, and a Linux/Python jerk ; the problem is most likely between my seat and my keyboard).

Thanks for your work : there is very little information about GoPro WiFi communications.

yet-another-average-joe, 2019-02-04:
Thanks ! It works !

I had a hard time installing and compiling Python 3.6.5 : failed twice and reflashed 3 times before I could invoque python3.6. 

This could be usefull to OctoPi/OctoPrint users : don't follow the steps from : https://gist.github.com/dschep/24aa61672a2092246eaca2824400d37f ; at the end typing python3.6 fails (no such file or directory). Instead, use https://liftcodeplay.com/2017/06/30/how-to-install-python-3-6-on-raspbian-linux-for-raspberry-pi/ ; and "make -j 3" really makes a huge difference on the multicore MCU. Foosel is preparing a new release, hope she will implement Python 3.6 in the distro. (I did'nt test the release candidates, so I don't know)

I also tested the 10 sec video example with Python 3.5 (typing python3 wich comes by default), and it worked.

Something weird : when typing "goproCamera = GoProCamera.GoPro()" with the GoPro (Hero3+ Black) OFF, I get lots of errors, but it turns ON. The video command fails. But retyping "goproCamera = GoProCamera.GoPro()", then it works fine. Still have to RTFM !

Your library is easier tu use than bare urllib, by far ! (at first, I tried and failed).

Virtual environments... I have to grow somewhat before I try.

Thanks again !

closed 2019-02-04 by KonradIT