KonradIT/gopro-py-api #83: Hero3+ Black and power_on

yet-another-average-joe, 2019-02-06:
Hello,
I can't wake up the GoPro Hero3+BE using power_on() or power_on_auth()

power_on(), with or without the MAC address says "waking up" and returns, but nothing happens.
power_on_auth() throws exceptions.

I've read issues #2 and #4, it did'nt help much. Do some models have secret commands ?

I have a remote, 1x Pi 3B+ but I don't know how to sniff communications while waking it up with the remote. What software or methodology did you use ? I imagine the RasPi has to be configured as a repeater ?

yet-another-average-joe, 2019-02-24:
I partially solved my problem :
- I deactivated onboard WiFi on my RPi 3B+
- I plugged a USB WiFi dongle, and configured it

Now, I can power_on_auth() and power_off() my GoPro 3 H3+BE. But the camera has to be ON before invoquing the GoPro contructor... 

If not, power_on_auth() returns lots of errors, but turns on. The gopro object seems to be destroyed. The contructor has to be called a 2nd time, and then these 2 functions can be used without any problems, as expected.

Onboard WiFi seems to have very poor performance or compatibility issues with the GoPro H3. This seems to be well known with some WiFi routers and AP. Could this be the explanation ? See #88

KonradIT, 2019-02-24:
Yes this is a known issue (having to issue constructor call x2)

closed 2019-02-24 by KonradIT