KonradIT/gopro-py-api #84: livestream stability issues

OlivierHartmann, 2019-02-08:
Hi,
Thank you very much for your api.

I have some stability issues with the livestream of my HERO 7 black. In fact, sometimes i can have the stream for 15-20 min, and then it can stop definitively. The gopro does not answer anymore, I can't even ping it or its buttons or screen does not act anymore ! I then have to reboot it.
More, during the stream, I have stream holes of few seconds.
The final view on VLC (after a packet size reduction on the ffmeg forwarding) has not a great quality and too many image loses and glitches.

Do you have any idea to improve all of that ?
Thank you again.
Olivier

KonradIT, 2019-02-08:
Which firmware are you on? I had some of those issues in earlier firmwares with the 7 Black but not with the newer ones.

OlivierHartmann, 2019-03-06:
Hi,
I am so sorry for the delay, I forgot to change my email address to get notifications.

My firmware version is 01.70