KonradIT/gopro-py-api #91: GoPro Hero Session Live Streaming quality

Phantasmiic, 2019-04-02:
Is there a max resolution for streaming from the **GoPro Hero Session**? Otherwise, how can I change it?

I have only been able to get 340 x 432. I have tried changing the Video.RESOLUTION parameter, and setting the quality parameter in the `stream()` function in the API. 

[image]

I am using Ubuntu 16, and am running `gopro_keepalive()` , and `gopro_opencv()`. 

Thanks!

takov751, 2019-06-19:
greetings as far as i know that is the maximum resolution for streaming,because its only used as a preview function, where you can check the target/composition when you want to record/capture

takov751, 2019-06-19:
greetings as far as i know that is the maximum resolution for streaming,because its only used as a preview function, where you can check the target/composition when you want to record/capture. This API has nothing to do with that.