KonradIT/gopro-py-api #92: KeepAlive don't work - GoPro Hero 6

tacleur, 2019-04-07:
Hi everybody,

Konrad, Thanks a lot for your Api !

I have some questions for my project :

The KeepAlive don't works and block my python code.

`#!/usr/bin/python3

# IMPORTATION

from gpiozero import Button
from time import sleep
from goprocam import GoProCamera, constants
import goprocam

goproCamera = GoProCamera.GoPro()
goproCamera.pair(usepin=False)
goproCamera.video_settings("2k",'24')
goproCamera.shoot_video(10)
goproCamera.shutter(constants.stop)
goproCamera.KeepAlive()
print('done')`

the result : 
`HERO6 Black
HD6.01.02.01.00
Camera successfully connected!
Paired
{}

{}

{}

{}

{}

`

KeepAlive paused my code.

What's the solution for KeepAlive ?

An other question : how can i change the FOV for my video ?

Thanks a lot for your answer

KonradIT, 2019-04-07:
This is because keepAlive needs to be executed on a separate thread.

tacleur, 2019-04-07:
Thanks a lot for your return, I have tried by i have a problem of thread in my boucle (while...) . I keep going.

How can i change the FOV (large...) on photo and video. I find how change FR and Resolution but how can i change the other parameter?

Thanks a lot

takov751, 2019-06-19:
As far as i remember fov can be changed just like the fps just with angle integer `fpv=90` 
And for the keepalive function, I think python threading modul could be a good choice to run it as a daemon thread, while the script running.