KonradIT/gopro-py-api #96: troubles with Hero 7 Black streaming

tylerstraub, 2019-05-26:
When running the .stream() command on the Hero 7 Black we are receiving a lot of problematic output. 

Any ideas out there? Has anyone been successful firing up a UDP stream on the Hero7 Black? 

Thanks!

**gopro.stream("udp://127.0.0.1:10000")**

ffmpeg version 3.4.6 Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 7.4.0 (OpenWrt GCC 7.4.0 r10074-c4abaa3)
  configuration: --enable-cross-compile --cross-prefix=aarch64-openwrt-linux-musl- --arch=aarch64 --cpu=cortex-a53 --target-os=linux --prefix=/usr --pkg-config=pkg-config --enable-shared --enable-static --enable-pthreads --enable-zlib --disable-doc --disable-debug --disable-lzma --disable-vaapi --disable-vdpau --disable-outdevs --enable-avresample --disable-decoder=atrac3 --disable-decoder=h264 --disable-decoder=hevc --disable-decoder=vc1 --disable-muxer=h264 --disable-muxer=hevc --disable-muxer=vc1 --disable-demuxer=h264 --disable-demuxer=hevc --disable-demuxer=vc1 --disable-parser=h264 --disable-parser=hevc --disable-parser=vc1 --enable-libopus --enable-small --enable-libmp3lame
  libavutil      55. 78.100 / 55. 78.100
  libavcodec     57.107.100 / 57.107.100
  libavformat    57. 83.100 / 57. 83.100
  libavdevice    57. 10.100 / 57. 10.100
  libavfilter     6.107.100 /  6.107.100
  libavresample   3.  7.  0 /  3.  7.  0
  libswscale      4.  8.100 /  4.  8.100
  libswresample   2.  9.100 /  2.  9.100
[h264_v4l2m2m @ 0x7f84637840] Could not find a valid device
[h264_v4l2m2m @ 0x7f84637840] can't configure decoder
[mpegts @ 0x7f84292520] Failed to open codec in avformat_find_stream_info
[h264_v4l2m2m @ 0x7f84637840] Could not find a valid device
[h264_v4l2m2m @ 0x7f84637840] can't configure decoder
[mpegts @ 0x7f84292520] Could not find codec parameters for stream 0 (Video: h264 ([27][0][0][0] / 0x001B), none): unspecified size
Consider increasing the value for the 'analyzeduration' and 'probesize' options
[mpegts @ 0x7f84292520] Could not find codec parameters for stream 2 (Unknown: none ([128][0][0][0] / 0x0080)): unknown codec
Consider increasing the value for the 'analyzeduration' and 'probesize' options
[mpegts @ 0x7f84292520] Could not find codec parameters for stream 3 (Audio: ac3 ([129][0][0][0] / 0x0081), 0 channels, fltp): unspecified sample rate
Consider increasing the value for the 'analyzeduration' and 'probesize' options
Input #0, mpegts, from 'udp://10.5.5.9:8554':
  Duration: N/A, start: 0.400400, bitrate: N/A
  Program 1 
    Stream #0:0[0x1011]: Video: h264 ([27][0][0][0] / 0x001B), none, 29.97 tbr, 90k tbn, 90k tbc
    Stream #0:1[0x1100]: Audio: aac ([15][0][0][0] / 0x000F), 48000 Hz, stereo, fltp, 192 kb/s
    Stream #0:2[0x200]: Unknown: none ([128][0][0][0] / 0x0080)
    Stream #0:3[0x201]: Audio: ac3 ([129][0][0][0] / 0x0081), 0 channels, fltp
Please use -b:a or -b:v, -b is ambiguous
[h264_v4l2m2m @ 0x7f832747e0] Could not find a valid device
[h264_v4l2m2m @ 0x7f832747e0] can't configure decoder
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (h264_v4l2m2m) -> mpeg2video (native))
  Stream #0:1 -> #0:1 (aac (native) -> mp2 (native))
Error while opening decoder for input stream #0:0 : Invalid argument

KonradIT, 2019-05-26:
This is running on the raspberry pi right? I'd suggest uninstalling FFmpeg and compiling FFmpeg from source (https://git.ffmpeg.org/ffmpeg.git)

tylerstraub, 2019-05-26:
It is, but the distribution is not a typical one (OpenMPTCPRouter, a derivative of OpenWRT). Perhaps the OPKG build of ffmpeg is too out of date. Thank you for pointing me in this direction, I will attempt to build FFMPEG from source and report back on my findings.

tylerstraub, 2019-05-28:
It was indeed FFMPEG. 

We were able to test the UDP feed but the overall stability and quality will not meet the requirements for what I'm looking for unfortunately. 

Going to have to use an external encoder after all. Thank you for pointing us in the right direction!

closed 2019-05-28 by tylerstraub