KonradIT/gopro-py-api #101: Initial setup issue (linux/python problem)

chessnecessary, 2019-06-17:
Following the README in gopro-py-api: I am up to the step which requires the command:
from goprocam import GoProCamera, constants

I believe this should be easy to troubleshoot, but my minimal experience with Linux and Python both is causing me trouble. I have already successfully installed everything instructed at the beginning of the document. Anytime I attempt the command, it seems to cause many errors and I'm not sure what's happening. Right now, my process is:

pi@LoRaGateway:~ $ cd gopro-py-api
pi@LoRaGateway:~/gopro-py-api $ python
Python 2.7.13 (default, Sep 26 2018, 18:42:22) 
[GCC 6.3.0 20170516] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from goprocam import GoProCamera, constants
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "goprocam/__init__.py", line 1, in <module>
    from goprocam import GoProCamera
  File "goprocam/GoProCamera.py", line 3, in <module>
    import urllib.request
ImportError: No module named request
>>>

Any help would be appreciated, I'm pretty positive the issue is simply stemming from my lack of know-how in linux and/or python.

yet-another-average-joe, 2019-06-17:
It seems you are using Python 2.7. gopro-py-api relies on Python 3.6 !

chessnecessary, 2019-06-17:
How would I install the updated version? Would it be related to the "python setup.py install" command?
Thank you for the swift response...

KonradIT, 2019-06-17:
Use python 3 and pip3

chessnecessary, 2019-06-17:
So not the linux terminal? Apologies for misunderstanding... I have the Python 3 (IDLE) program installed already, so should I run the commands from there instead?

chessnecessary, 2019-06-17:
or simply use linux commands to install pip3 and python3 instead

yet-another-average-joe, 2019-06-17:
Maybe this could help the OP. I made a video about installing gopro-py-api and Python 3.6

There are scripts that make things easy (in the description).

https://www.youtube.com/watch?v=TML8L3RmN60

chessnecessary, 2019-06-17:
Everything seemed to install fine, but now I get this error, and the "python3.6" command will not function. It seems a website is not functioning?

This will install Python 3.6. Do you want to proceed (Y/N)? 
Installing Python 3.6
Installing Python 3.6...

Get:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Hit:2 http://archive.raspberrypi.org/debian stretch InRelease
Fetched 15.0 kB in 1s (10.4 kB/s)
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version (12.3).
checkinstall is already the newest version (1.6.2-4).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libbz2-dev is already the newest version (1.0.6-8.1).
libc6-dev is already the newest version (2.24-11+deb9u4).
libgdbm-dev is already the newest version (1.8.3-14).
libncursesw5-dev is already the newest version (6.0+20161126-1+deb9u2).
libreadline-gplv2-dev is already the newest version (5.2+dfsg-3).
libsqlite3-dev is already the newest version (3.16.2-5+deb9u1).
libssl-dev is already the newest version (1.1.0j-1~deb9u1).
tk-dev is already the newest version (8.6.0+9).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
--2019-06-17 12:53:07--  http://wget/
Resolving wget (wget)... failed: Name or service not known.
wget: unable to resolve host address ‘wget’
--2019-06-17 12:53:07--  https://www.python.org/ftp/python/3.6...
Resolving www.python.org (www.python.org)... 151.101.0.223, 151.101.64.223, 151.101.128.223, ...
Connecting to www.python.org (www.python.org)|151.101.0.223|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2019-06-17 12:53:07 ERROR 404: Not Found.

tar (child): Python-3.6.0.tgz: Cannot open: No such file or directory
tar (child): Error is not recoverable: exiting now
tar: Child returned status 2
tar: Error is not recoverable: exiting now
./install-python3.6.sh: line 58: cd: Python-3.6.0: No such file or directory
bash: configure: No such file or directory
make: *** No rule to make target 'altinstall'.  Stop.

test launch Python 3.6 : python3.6 command
type exit() to quit Python3.6

yet-another-average-joe, 2019-06-17:
OOPS !!!

I think you followed the instructions in the tutorial. And I made a BIG mistake !!!

I wrote (in the script) :

```
# donwnload Python 3.6 sources archive
wget wget https://www.python.org/ftp/python/3.6.0/Python-3.6.0.tgz
```

wget wget !

delete one wget !

I edit the comments right now !

(I don't understand how this came in the script ; I tested this procedure maybe 20 or 30 times before I released ; but I'm also a beginner...)

chessnecessary, 2019-06-17:
I got Python3.6.0 installed! Now, I have reached a new hurdle:

After defining my camera as the function "GoProCamera.GoPro()", it gives an extremely (extremely) long string of 2 repeating lines:

HERO6 Black
HD6.01.02.01.00

until finally giving a RecursionError in python and shutting it down, leaving it still undefined...
Any initial thoughts?

chessnecessary, 2019-06-17:
This means that it is at least recognizing the gopro for what it is while connected to its wi-fi signal! But no commands using the GoPro's defined name work, as the variable never completes definition...

yet-another-average-joe, 2019-06-17:
At this point, I can't help, as I have no H6. But have a look to the constructor in gopro-py-api, AFAIR it could depend on the GoPro version. Many things changed since the H3. And remember that the onboard WiFi sucks. On my side, I had many issues until I use a WiFi USB dongle.

KonradIT, 2019-06-17:
A RecursionError? What's your code?

chessnecessary, 2019-06-17:
pi@LoRaGateway:~ $ cd gopro-py-api
pi@LoRaGateway:~/gopro-py-api $ python3.6
Python 3.6.0 (default, Jun 17 2019, 13:16:50) 
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from goprocam import GoProCamera, constants
>>> LoftCam2 = GoProCamera.GoPro()

  HERO6 Black
  HD6.01.02.01.00      

 *repeats for about 30sec*

File "/home/pi/gopro-py-api/goprocam/GoProCamera.py", line 235, in infoCamera
    if self.whichCam() == constants.Camera.Interface.GPControl:
  File "/home/pi/gopro-py-api/goprocam/GoProCamera.py", line 154, in whichCam
    self.prepare_gpcontrol()
  File "/home/pi/gopro-py-api/goprocam/GoProCamera.py", line 23, in prepare_gpcontrol
    firmware = self.infoCamera(constants.Camera.Firmware)

*repeats for about 30sec*

  File "/home/pi/gopro-py-api/goprocam/GoProCamera.py", line 235, in infoCamera
    if self.whichCam() == constants.Camera.Interface.GPControl:
  File "/home/pi/gopro-py-api/goprocam/GoProCamera.py", line 154, in whichCam
    self.prepare_gpcontrol()
  File "/home/pi/gopro-py-api/goprocam/GoProCamera.py", line 23, in prepare_gpcontrol
    firmware = self.infoCamera(constants.Camera.Firmware)
  File "/home/pi/gopro-py-api/goprocam/GoProCamera.py", line 235, in infoCamera
    if self.whichCam() == constants.Camera.Interface.GPControl:
  File "/home/pi/gopro-py-api/goprocam/GoProCamera.py", line 154, in whichCam
    self.prepare_gpcontrol()
  File "/home/pi/gopro-py-api/goprocam/GoProCamera.py", line 23, in prepare_gpcontrol
    firmware = self.infoCamera(constants.Camera.Firmware)
  File "/home/pi/gopro-py-api/goprocam/GoProCamera.py", line 235, in infoCamera
    if self.whichCam() == constants.Camera.Interface.GPControl:
  File "/home/pi/gopro-py-api/goprocam/GoProCamera.py", line 140, in whichCam
    response_raw = urllib.request.urlopen('http://' + self.ip_addr + '/gp/gpControl', timeout=5).read().decode('utf8')
  File "/usr/local/lib/python3.6/urllib/request.py", line 223, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/local/lib/python3.6/urllib/request.py", line 526, in open
    response = self._open(req, data)
  File "/usr/local/lib/python3.6/urllib/request.py", line 544, in _open
    '_open', req)
  File "/usr/local/lib/python3.6/urllib/request.py", line 504, in _call_chain
    result = func(*args)
  File "/usr/local/lib/python3.6/urllib/request.py", line 1346, in http_open
    return self.do_open(http.client.HTTPConnection, req)
  File "/usr/local/lib/python3.6/urllib/request.py", line 1318, in do_open
    encode_chunked=req.has_header('Transfer-encoding'))
  File "/usr/local/lib/python3.6/http/client.py", line 1239, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/local/lib/python3.6/http/client.py", line 1285, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/usr/local/lib/python3.6/http/client.py", line 1234, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/local/lib/python3.6/http/client.py", line 1026, in _send_output
    self.send(msg)
  File "/usr/local/lib/python3.6/http/client.py", line 964, in send
    self.connect()
  File "/usr/local/lib/python3.6/http/client.py", line 936, in connect
    (self.host,self.port), self.timeout, self.source_address)
  File "/usr/local/lib/python3.6/socket.py", line 704, in create_connection
    for res in getaddrinfo(host, port, 0, SOCK_STREAM):
  File "/usr/local/lib/python3.6/socket.py", line 745, in getaddrinfo
    addrlist.append((_intenum_converter(af, AddressFamily),
  File "/usr/local/lib/python3.6/socket.py", line 103, in _intenum_converter
    return enum_klass(value)
  File "/usr/local/lib/python3.6/enum.py", line 291, in __call__
    return cls.__new__(cls, value)
  File "/usr/local/lib/python3.6/enum.py", line 519, in __new__
    if type(value) is cls:
RecursionError: maximum recursion depth exceeded while calling a Python object

chessnecessary, 2019-06-17:
This is all while the board has been keeping (seemingly) constant connection to the camera and the internet

KonradIT, 2019-06-17:
Sorry, it was [this PR](https://github.com/KonradIT/gopro-py-api/pull/98/files). Perform a `git pull`.

Code needs some cleaning, including our own exception, utils, private functions, logger, tests.... I made this when my python knowledge was not very good :)

chessnecessary, 2019-06-17:
Could you define "git pull" haha. Where should I input this command?
Glad you were able to spot the issue, thank you!

KonradIT, 2019-06-17:
In the command line.

chessnecessary, 2019-06-17:
pi@LoRaGateway:~ $ cd gopro-py-api
pi@LoRaGateway:~/gopro-py-api $ git pull
Already up-to-date.
pi@LoRaGateway:~/gopro-py-api $ python3.6
Python 3.6.0 (default, Jun 17 2019, 13:16:50)
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.

from goprocam import GoProCamera, constants
LoftCam2 = GoProCamera.GoPro()

Then the same issue occurs. Do I need to do something else to update the api from github?

chessnecessary, 2019-06-17:
(ignore the cross-throughs, not sure why those appeared)

chessnecessary, 2019-06-17:
It successfully updated, and successfully recorded the 10sec test footage. Thank you for your help! I may be back soon for more--my goal is currently to transmit photos over LoRa using the GoPro.

closed 2019-06-19 by chessnecessary