KonradIT/gopro-py-api #103: Trouble Connecting to WiFi

chessnecessary, 2019-06-19:
I realize that this issue is not directly related to the gopro api, but nonetheless, I thought that this would be the most responsive community in helping me find a tutorial/answer to the issue which actually works!

I've gotten everything I've tested within the gopro api to work so far on a "normal" raspberry pi 3, and am now attempting to do the same on my 2 Pi Zeros. The issue I'm running into is that I cannot connect to wifi with them. They have the capability, I just cannot seem to get them to connect to any wifi network (including the GoPro). 

A primary contributor to this issue is that I'm attempting to connect to wifi using only the Linux terminal, as I do not seem to have access to a Windows-like interface (possibly due to the fact that these are Pi-In-The-Sky Pi Zeros, intended for use on balloons and shipped with a pre-built SD card).

Any help would be great, thank you in advance!

KonradIT, 2019-06-19:
What is your GoPro model? Good chance it cannot connect to it because the GoPro is on 5GHz WiFi and the Pi Zero W [supports 2.4GHz only](https://www.digikey.com/catalog/en/partgroup/bcm43143-and-bcm43362/62925).

On your GoPro you can change the frequency to 2.4GHz

chessnecessary, 2019-06-19:
It's already on 2.4Ghz (and the GoPro is a HERO6, btw). The Pi Zero won't connect to any WiFi so I think that the issue may be elsewhere.

chessnecessary, 2019-06-20:
Any help is appreciated!

closed 2019-06-26 by chessnecessary