KonradIT/gopro-py-api #104: from goprocam import GoProCamera, constants

ZIKO94ZIKO, 2019-06-21:
Please i need a help , I would connect gopro hero 5 whith Raspbery pi 3 , when i tap command : python , i get that message 
**Screenshots**
from goprocam import GoProCamera, constants
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "goprocam/__init__.py", line 1, in <module>
    from goprocam import GoProCamera
  File "goprocam/GoProCamera.py", line 3, in <module>
    import urllib.request
ImportError: No module named request

KonradIT, 2019-06-21:
Use Python 3.6 or newer

closed 2019-06-21 by KonradIT

ZIKO94ZIKO, 2019-06-22:
Done :D Thank u very much for your help , i have just a question : i want to do stream live with GO PRO HERO 5  and save data in Raspbery pi 3 (cloud) ,can u help me please.