KonradIT/gopro-py-api #105: HERO7 Black No Recording / Shutter

sensor-dev, 2019-06-23:
Unable to initiate the shutter command. 
Connection works fine and I can read settings and media information but I'm unable to record pictures or videos.
While debugging I see the following commands beeing sent:
- http://10.5.5.9/gp/gpControl/command/sub_mode?mode=1&sub_mode=0
- http://10.5.5.9/gp/gpControl/command/shutter?p=1

The first one works and the camera goes into Photo mode. The second one returns: 
- **HTTP Error 500: Internal Server Error**

Manualy executing the url in the browser returns:
- **SyntaxError: JSON.parse: bad control character in string literal at line 6 column 40 of the JSON data**

Additional Info:
- camera model: **HERO7 Black**
- firmware version: **HD7.01.01.70.00**

Any idea?

KonradIT, 2019-06-23:
Weird, that's definitely something wrong with the camera. I'll try to repro on my H7 Black later.

sensor-dev, 2019-06-23:
Ok, I think I found the problem. I didn't pair my GoPro with the computer.
After the first connection, I had to perform a pairing with the following command:
- http://10.5.5.9/gp/gpControl/command/wireless/pair/complete?success=1&deviceName=DESKTOP

As I didn't use the desktop app, the pairing was not performed. 
This step should be documented a bit more visible. As the other commands were successful, I didn't think this was necessary :)

KonradIT, 2019-06-23:
The API assumes the camera is paired beforehand.

To pair a new camera without using the smartphone app:

````python

gpCam = GoProCamera.GoPro()
gopro.pair(usePin=False)
````

closed 2019-06-23 by sensor-dev