KonradIT/gopro-py-api #106: Fusion1 only streams back camera

BlueHorn07, 2019-06-24:
This code works well on taking photos and shooting video. For streaming, there's some troubles to operate(ex: ffmpeg) but, I do the task! But, I got new problem about streaming on GoPro Fusion1.

my program streams only back camera!! ( GoPro fusion1 has two cameras, front & back )

I do closely look at the whole code but, I wasn't able to find what's wrong with it.

I made two thread with different udp url and start them, but it doesn't work!!

I know u update some of features of fusion (ex: getMediafusion())
But, I can't handle this streaming problem.

I think key problem is in livestream(self) method 

   
`	def livestream(self,option): 

		"""start livestreaming
		option = "start"/"stop"
		"""
		if option == "start":
			**if self.whichCam() == "gpcontrol":
				print(self.gpControlExecute('p1=gpStream&a1=proto_v2&c1=restart'))**
			else:
				print(self.sendCamera("PV","02"))	
		if option == "stop":
			if self.whichCam() == "gpcontrol":
				print(self.gpControlExecute('p1=gpStream&a1=proto_v2&c1=stop'))
			else:
				print(self.sendCamera("PV","00"))
`
'p1=gpStream&a1=proto_v2&c1=restart' 
I think back camera is set as default. and I guess that there's more options so that i can access Front Camera. 
Because it send command to gpControl/execute place. i think "p1=gpStream" or "a1=proto_v2" should be changed to support both camera streaming feature!!

`
	def gpControlExecute(self, param):

		"""sends Parameter to gpControl/execute"""
		try:
			return urllib.request.urlopen('http://' + self.ip_addr + '/gp/gpControl/execute?' + **param**, timeout=5).read().decode('utf-8')
		except (HTTPError, URLError) as error:
			return ""
		except timeout:
			return ""`

Anyone has an idea of streaming both front and back camera of fusion1, please share it...

KonradIT, 2019-06-24:
This one is a bit tricky, there's nothing wrong with the existing code, it's just that `10.5.5.9:8554` sends the two streams together!

To access 1 stream (1 is back, 2 is front):

````
from goprocam import GoProCamera, constants
gopro = GoProCamera.GoPro()
gopro.livestream("start")
gopro.KeepAlive()
````

Then using MPV:

Front: `mpv udp://10.5.5.9:8554 --vid=2`
Back: `mpv udp://10.5.5.9:8554 --vid=1`

I need to find a way to display the two stream superimposed or next to each other, no idea about 360 streaming, but if you find how it works and the specificactions needed let me know.

Specs for the stream: (h264 736x720 29.970fps) (both front and back).

BlueHorn07, 2019-06-25:
Thank you for your comment :)
Now, I can get front streaming!! But, If I'm streaming front, the back one didn't work.

[image]
(upper one: vid=1, lower one: vid=2

the error messages were
[ffmpeg/demuxer] mpegts: Could not find codec parameters for stream 2 (Unknown: none ([128][0][0][0] / 0x0080)): unknown codec
[ffmpeg/demuxer] Consider increasing theff value for the 'analyzeduration' and 'probesize' options.

KonradIT, 2019-06-25:
You cannot have 2 MPV windows of the same stream at the same time. You'd need to transcode 1 stream to a different port.

BlueHorn07, 2019-06-25:
I solved this problem by using ffmpeg option.

In shell(anaconda)

1. 
`ffmpeg -i udp://10.5.5.9:8554  -map 0:v:0 -f mpegts ./back.mp4  -map 0:v:1 -f mpegts ./front.mp4`
2.
`mpv ./back.mp4`
3. 
`mpv ./front.mp4`

In code
```
    def streaming(self):
        self.gopro.livestream("start")
        self.gopro.KeepAlive()
        time.sleep(0.5)
```

```
        url = ('udp://10.5.5.9:8554')
        BackCam_File = ('back_' + str(time.time()) + '.mp4')
        FrontCam_File = ('front_' + str(time.time()) + '.mp4')
        cam_stream = threading.Thread(target = self.streaming)
        cam_stream.start()

        time.sleep(1)
        stream = subprocess.Popen('ffmpeg -i '+ url + ' -map 0:v:0 -f mpegts ./' + BackCam_File + ' -map 0:v:1 -f mpegts ./' + FrontCam_File, shell=True)
        time.sleep(10)
        back = subprocess.Popen('mpv ./' + BackCam_File, shell=True)
        front = subprocess.Popen('mpv ./'+  FrontCam_File, shell=True)
```