KonradIT/gopro-py-api #107: Issue running simple python script

chessnecessary, 2019-06-27:
I am attempting to ultimately set up a script that will run upon boot of my raspberry pi (so that it can function headlessly). However, I currently cannot even get it to run the script manually. Here is the script: 

#! /usr/src python3.6
from goprocam import GoProCamera, constants
goproCamera = GoProCamera.GoPro()
goproCamera.shoot_video(10)

and to run it, I am using: sudo python3.6 gopro-py-api/goproboot.pi
  while in the home/pi default folder.

I will post the error messages I am receiving in a few minutes, I am confused as to why the script will not run at all.

chessnecessary, 2019-06-27:
[image]

KonradIT, 2019-06-27:
You have multiple errors in your script, 

>#! /usr/src python3.6

should be `#!/usr/bin/python3.6` or `#!/usr/src/python3.6`

The file should be called `goproboot.py` not `goproboot.pi`

KonradIT, 2019-06-27:
Ah I see the picture now, do you have your camera with WiFi running and connected to your Pi?

chessnecessary, 2019-06-27:
The "goproboot.pi" was a typo in the message, I'll change the beginning of the script to reflect what you said though.
Yes it is, and if I run the code manually (outside of the script) it works fine.

KonradIT, 2019-06-27:
Because the code runs at boot maybe it runs before the Pi can connect to the camera?

chessnecessary, 2019-06-27:
The errors are occurring when I run it post-login. I gave up on having it run at boot until I can get it working with manual prompting. Those errors accumulated in the picture are from a manual attempt.

chessnecessary, 2019-06-27:
Ignore me, I had the passcode incorrect. I will update shortly if an issue persists but when I changed the wpa files from my wireless internet connection back to the GoPro I entered in an incorrect character.

chessnecessary, 2019-06-27:
It's actually giving me the same errors, so it seems to be an issue with connection to the GoPro, not the script at the moment. I haven't changed anything as far as connectivity goes on the GoPro or the Pi Zero W in the last week, and they worked fine together on Monday. Any clues given from the errors?

KonradIT, 2019-06-27:
As I said before, you need to ensure the script runs after the Pi has connected to WiFi and can access 10.5.5.9:8080, a simple ping script can do that.