KonradIT/gopro-py-api #116: Problem on connecting GoProHero4

viet2411, 2019-10-03:
Hello,
I'm trying to connect GoPro Hero 4 to my laptop by using your code,but it doesn't work even with simple code. Can you tell me the mistakes?
I used this code:
`from goprocam import GoProCamera, constants
goproCamera = GoProCamera.GoPro()
goproCamera.shoot_video(10)`
and got this error
`Waking up...
Waking up...
Traceback (most recent call last):
  File "/usr/lib/python3.6/urllib/request.py", line 1318, in do_open
    encode_chunked=req.has_header('Transfer-encoding'))
  File "/usr/lib/python3.6/http/client.py", line 1254, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/lib/python3.6/http/client.py", line 1300, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.6/http/client.py", line 1249, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.6/http/client.py", line 1036, in _send_output
    self.send(msg)
  File "/usr/lib/python3.6/http/client.py", line 974, in send
    self.connect()
  File "/usr/lib/python3.6/http/client.py", line 946, in connect
    (self.host,self.port), self.timeout, self.source_address)
  File "/usr/lib/python3.6/socket.py", line 704, in create_connection
    for res in getaddrinfo(host, port, 0, SOCK_STREAM):
  File "/usr/lib/python3.6/socket.py", line 745, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
socket.gaierror: [Errno -2] Name or service not known

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/vietviet/.local/lib/python3.6/site-packages/goprocam/GoProCamera.py", line 152, in sendCamera
    self.getPassword() + value_notempty)
  File "/home/vietviet/.local/lib/python3.6/site-packages/goprocam/GoProCamera.py", line 112, in _request
    return urllib.request.urlopen(uri, timeout=_timeout, context=_context).read().decode("utf-8")
  File "/usr/lib/python3.6/urllib/request.py", line 223, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python3.6/urllib/request.py", line 526, in open
    response = self._open(req, data)
  File "/usr/lib/python3.6/urllib/request.py", line 544, in _open
    '_open', req)
  File "/usr/lib/python3.6/urllib/request.py", line 504, in _call_chain
    result = func(*args)
  File "/usr/lib/python3.6/urllib/request.py", line 1346, in http_open
    return self.do_open(http.client.HTTPConnection, req)
  File "/usr/lib/python3.6/urllib/request.py", line 1320, in do_open
    raise URLError(err)
urllib.error.URLError: <urlopen error [Errno -2] Name or service not known>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "untitled0.py", line 4, in <module>
    gopro.stream("udp://127.0.0.1:10000")
  File "/home/vietviet/.local/lib/python3.6/site-packages/goprocam/GoProCamera.py", line 1155, in stream
    self.livestream("start")
  File "/home/vietviet/.local/lib/python3.6/site-packages/goprocam/GoProCamera.py", line 1143, in livestream
    return self.sendCamera("PV", "02")
  File "/home/vietviet/.local/lib/python3.6/site-packages/goprocam/GoProCamera.py", line 154, in sendCamera
    print("Error code:" + str(error.code) +
AttributeError: 'URLError' object has no attribute 'code'
`

yanesha, 2019-10-04:
I have a same problem in Python 3.7

atorres-eqrx, 2020-12-07:
I'm having the same issue. Hero Session. Python 3.6.12

omitheaussie, 2021-04-12:
I have the same problems python 3.7.7