KonradIT/gopro-py-api #117: Livestream error

silas229, 2019-10-26:
**Describe the bug**
I get this error when I execute [examples/livestreaming/gopro-keepalive.py](https://github.com/KonradIT/gopro-py-api/blob/master/examples/streaming/gopro_keepalive.py):
```sh
$ gopro-keepalive.py
HERO5 Black
HD5.02.02.70.00
Camera successfully connected!
ffmpeg version N-94664-g0821bc4eee Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 9.1.1 (GCC) 20190807
  configuration: --enable-gpl --enable-version3 --enable-sdl2 --enable-fontconfig --enable-gnutls --enable-iconv --enable-libass --enable-libdav1d --enable-libbluray --enable-libfreetype --enable-libmp3lame --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopenjpeg --enable-libopus --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libtheora --enable-libtwolame --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxml2 --enable-libzimg --enable-lzma --enable-zlib --enable-gmp --enable-libvidstab --enable-libvorbis --enable-libvo-amrwbenc --enable-libmysofa --enable-libspeex --enable-libxvid --enable-libaom --enable-libmfx --enable-amf --enable-ffnvcodec --enable-cuvid --enable-d3d11va --enable-nvenc --enable-nvdec --enable-dxva2 --enable-avisynth --enable-libopenmpt
  libavutil      56. 33.100 / 56. 33.100
  libavcodec     58. 55.101 / 58. 55.101
  libavformat    58. 31.104 / 58. 31.104
  libavdevice    58.  9.100 / 58.  9.100
  libavfilter     7. 58.101 /  7. 58.101
  libswscale      5.  6.100 /  5.  6.100
  libswresample   3.  6.100 /  3.  6.100
  libpostproc    55.  6.100 / 55.  6.100
[udp @ 000002c01ef5a100] bind failed: Error number -10048 occurred
udp://:8554: I/O error
```

Guatama, 2019-11-11:
I have also problem with livestream. After that command just nothing happen (and gopro-opencv.py failed too). 
[image]

ykzk, 2019-12-18:
Try disabling your firewall.

Capricorn-3th, 2021-01-23:
I have same problem,and fix it by reboot computer