KonradIT/gopro-py-api #121: Hero 8

Chick92, 2019-11-13:
Is the Hero 8 Black edition supported?

KonradIT, 2019-11-13:
Soon :)

KonradIT, 2019-11-13:
To clarify, most stuff works, just have to get the specific settings working. Try it out and let me know.

Chick92, 2019-11-13:
I've not bought one yet. 
Was going to buy a hero & as your API supports it. But the Hero 8 has better image stabilisation and mounts, both of which are required. 
When you say most stuff works, I just need to record 4K in low light conditions with the image stabilisation - does that work yet? 
I'm happy to contribute if I can.

KonradIT, 2019-11-13:
I'm waiting for GoPro to send over a H8 to try, I'll push an update this weekend that adds some MAX-specific settings. What I mean is, most API functions will work (change settings, start/stop recording, take pictures,....) but some settings are missing. Getting media off the camera should work as well, same for streaming (not 100% verified still).

Chick92, 2019-11-13:
Awesome, thanks!

torolito, 2019-11-25:
I'm currently working with goprocam and a Hero6, Hero8 and a Max.
one thing I had to change was:
gpCam = GoProCamera.GoPro(constants.Camera.Interface.GPControl)

After this change it worked perfetly fine.

KonradIT, 2019-11-25:
Nice, I have the MAX and already updated the API to support it, does `GoProCamera.GoPro()` not work?

torolito, 2019-11-25:
GoProCamera.GoPro() did not work.
Somehow it gave me back an empty string and did print “waking up…”

 

 

I can’t remember where it hung up exactly but once I figured out that the json parsing worked manually, I’ve tried to force the gpcontrol and then it worked.

If you want I can check where I think the problem is.

torolito, 2019-11-28:
I think I've found it. The Hero8 is H19, will check later tonight, if I remember right.

GoProCamera.py
Line 190
`                exceptions = ["HX", "FS", "HD3.02", "H18"]`

hugangyongxiangsha, 2020-07-18:
> To clarify, most stuff works, just have to get the specific settings working. Try it out and let me know.

I've not bought 8，i want to get 1080p stream ，but stream or streamSettings function without effect，i only get 864*480