KonradIT/gopro-py-api #122: GoPro HERO8 / MAX / API refactor

KonradIT, 2019-11-14:
This weekend I'll push 4.1.0 which adds support for GoPro MAX commands, refactors the media download/get part to work better and more consistently across all cameras. GoPro HERO8 support is also in the works. 

Regarding turning WiFi on with MAX and HERO8, this will come in a later release as I need to figure out a way to interface with the computer BT chip across all major OSes asynchronously without requiring sudo access.

Peace.

torolito, 2019-11-22:
Awesome!
Looking forward to your BT solution, I didn't want do go down that rabbit hole.

Cheers

cwerner, 2019-11-28:
Great! Awaiting Hero8 compatibility ;-) 👍

mrgard, 2019-11-30:
Hi Konrad, today I tried the bluetooth commands on a HERO8, it works with bluetoothctl (scan on, trust, pair, connect, disconnect) and gatttool (connect).

However, when I try to issue commands for e.g. turn on wifi, nothing happens 
(char-write-req 2f 03170101). Maybe this is part of the different new commands on HERO8?

Also, connection with bluetoothctl and gatttool ONLY works when HERO8 pairing mode is active (when the GOPRO APP connect dialog is active). As soon as exiting this dialog on the camera itself, all bluetooth commands fails.

If you have any commands I can test, I'm more than happy to try :)

I'm testing from either a Win10 PC, or from a Raspberry PI4. 
Streaming works fine with low latency!

Keep up the good work!
greetings from Gard

eztutnahi, 2020-03-18:
Hi Konrad,
I'm trying to get control over a gopro Max but, though I managed to connect via wifi and start recording, once Max's recording i goes deaf and cannot stop it; I'm guessing it has to do with bluetooth connection (couldn't pair it). I've tried to upgrade goprocam from 4.0.2 to 4.1.0 (I assume that's were my problem is), but can't seem to be able to find the way to do it. I'm an absolute new to github, so I'm sure there's something wrong I'm doing. Could you give me guidance?
Working on Raspberry Pi3.
Thanks for the great work

Fougl, 2020-06-01:
> Hi Konrad,
> I'm trying to get control over a gopro Max but, though I managed to connect via wifi and start recording, once Max's recording i goes deaf and cannot stop it; I'm guessing it has to do [...]

@eztutnahi how did you manage to connect via wifi? My windows does not discover gopro max over wifi.

eztutnahi, 2020-06-02:
Hi @Fougl 
I believe that gopros activate wifi via bluetooth command. I was never capable of doing it with my raspi, though I managed to pair it. I configured the wifi in my raspi ([https://www.raspberrypi.org/documentation/configuration/wireless/wireless-cli.md]) and, whenever I want access to it I make a syncing attempt with the gopro phone app. Not the best solution, because I want to be able to control it remotely, but once connected to the gopro wifi, it'll keep connected no matter what, as long as the pi/gopro is not restarted.
hope it helps... a bit

KonradIT, 2020-11-21:
Well this has been going on for more than a year, as a progress update:

- Hero8 and MAX require WiFi to be turned on via a BLE command, see [gopro-ble-py](http://github.com/konradit/gopro-ble-py).
- Hero9 as well, but now the default USB behavior exposes the gpControl APIs, gpMediaList and new gpWebcam endpoints over USB Ethernet, so no WiFi or BLE fidgeting needed.
- This API now supports most Hero8 and Hero9 settings.