KonradIT/gopro-py-api #123: Not supported while recording or processing media

zongoalbert, 2019-11-25:
I am using gopro Hero 5 Black.
Everytime I connect and capture it works well . after 1 week of running it's can't be download the media , downloadLastMedia() option or downloadMedia() it shows "Not supported while recording or processing media."

The self.isRecording() functions returns 1.
can you explain this issue ,and how can be resolve it ? 
Thanks in advance,
Best Regard.

keith204, 2020-01-17:
same exact issue w/ GoPro Hero 5 Black.  Following.

gfleetwood, 2020-12-05:
This [comment](https://github.com/KonradIT/gopro-py-api/issues/10#issuecomment-731655229) might be useful. Seems like a version problem.

KonradIT, 2020-12-08:
Hi, I cannot reproduce this issue, my camera never returns 1 to IsRecording, but I did encounter another bug, which I have described here: https://github.com/KonradIT/gopro-py-api/issues/10#issuecomment-740778804