KonradIT/gopro-py-api #126: KeepAlive with GoPro Fusion

Falimonda, 2019-12-09:
KeepAlive does not seem to be working for the GoPro Fusion.

I have successfully connected to the Fusion using goprocam - I can capture, query media, etc.

Can anyone confirm if they've managed to keep their Fusion from shutting down after ~7 minutes using the KeepAlive method?

KonradIT, 2019-12-09:
I don't have a fusion anymore but when I had it KeepAlive worked fine...

Falimonda, 2019-12-09:
Thanks for the response, Konrad. I've just run Wireshark and see that the Fusion responds with "Destination/port unreachable" when sending the keep-alive message on port 8554. Maybe they've since changed the port for this feature.

I will poke around to see if I can find the right port for this. If all else fails, I will just trigger capture periodically as that also keeps it alive.

Falimonda, 2020-01-07:
Is there a suggested method to verify that the KeepAlive port is still 8554? The results of 'nmap -sU -p 8554 10.5.5.9' show that the port is closed. Maybe that's not indicative of much, considering UDP scan results on other ports give inconsistent results, but given that it takes ~7 minutes to verify whether the Fusion responds to my modified version of the KeepAlive function, a brute force approach is impractical and I'd really appreciate any guidance!