KonradIT/gopro-py-api #128: Project deployment issue

Awais9, 2020-01-05:
Hi there!
I wanted to deploy the code on Heroku with Flask server but it returns Application error. Please check the code and help me out on it. Please let me know if any detail is required I would be available.

KonradIT, 2020-01-05:
Uh, what? This isn't a server. Can you post the code?

KonradIT, 2020-01-05:
Also, Heroku can't connect to your GoPro, how do you even plan on connecting to the camera?

Awais9, 2020-01-05:
Yes, I know it is not server. I'm planning to deploy the code on Heroku and connect it with any instance by Shell so the front any could be any to work with. Does it will work with this adapter? Also provide me link I'll share the code. :)

Awais9, 2020-01-05:
I'm using python3 for it.

KonradIT, 2020-01-05:
This API needs an active connection to the camera. In the past people have used a different IP than `10.5.5.9` to connect to their cameras, for instance, if they're using a WLAN repeater. But I can't see why anyone would deploy this API in a Heroku server and expect it to work.

Awais9, 2020-01-05:
I'm planing to deploy RTMP server instance with it for active connection. I guess it should work. If there is any other possible way please let me know.

KonradIT, 2020-01-05:
Ok, so the RTMP server receives the live feed from a HERO7/8/MAX camera right?

Awais9, 2020-01-06:
Yes, so any suggestions would be appreciated. :)

KonradIT, 2020-01-06:
Suggestions for what exactly? I'm having a hard time understanding what you're trying to achieve by deploying this API wrapper in a Heroku instance, and why it's got anything to do with the RTMP server. This API *does not even support* setting the RTMP URL on the GoPro, this is done via Bluetooth. Closing as your issue is irrelevant to this project.

closed 2020-01-06 by KonradIT